ros-industrial/universal_robot #430: Update MoveIt configurations

gavanderhoorn, 2019-06-12:
Even though the provided MoveIt configurations are only there for convenience and quick testing of new robot setups (ie: right after unboxing) and are not intended for longer use (as any real application would need a customised MoveIt configuration), they should be up-to-date with the latest MoveIt changes.

This includes planner configurations, parameter and launch file names and structure of the package.

Rerunning the MSA and regenerating the packages would be a good first step, but care must be taken to not accidentaly override important configuration or remove settings that are needed for the correct functioning of MoveIt and UR robots.

gavanderhoorn, 2019-06-12:
Related: #431.

jacknlliu, 2019-06-14:
@gavanderhoorn just do it, please!

gavanderhoorn, 2019-06-18:
@jacknlliu: these issues were added in preparation for World ROS-I Day 2019.

Note the `support level: community` in the readme of this repository.

If you participate, perhaps you could even do it!

jeritgeorge, 2019-07-02:
I can work on this

gavanderhoorn, 2019-07-02:
@jeritgeorge: nice.

Please first investigate the current layout of the package. There are a few non-standard things that the UR moveit packages have.

It might be insightful to generate a completely new MoveIt configuration in a separate location and then comparing that to the current ones.

gavanderhoorn, 2019-07-03:
Did you make any progress here @jeritgeorge?

jeritgeorge, 2019-07-03:
Yes, I've reran the MSA for the ur3 config and went through and compared it to the current one. I think I have a decent understanding of the current setup and should at least have the ur3 config done either tonight or tomorrow. I think once one is done, the rest should be relatively simple.

gavanderhoorn, 2019-07-17:
I'm going to unassign this @jeritgeorge.

You're welcome to keep working on it of course.

fmauch, 2022-07-21:
I''l mark this closed, as #538 is merged.

closed 2022-07-21 by fmauch