ros-industrial/universal_robot #434: Connection with UR10e real

Drikuy, 2019-06-20:
I am testing connection between ROS and a UR10e real robot. I've followed the tutorial in order to launch the communication with the robot and moveit config but it does not work. And I have no idea about what can I do to fix it. Here I leave a picture of the rqt_graph. 

Thanks for your help.
[image]

gavanderhoorn, 2019-06-20:
You don't tell us anything about which packages you are using, but from your `rqt_graph` screenshot it would appear to be `ur_driver`.

As mentioned in the various tutorials and the wiki: you cannot use `ur_driver` with anything newer than `1.8.x`, so CB3 robots are not supported. e-series robots would also not be supported by that package.

The `kinetic-devel` branch of [ur_modern_driver](https://github.com/ros-industrial/ur_modern_driver/tree/kinetic-devel) does support CB3 controllers.

But e-series support is currently not integrated into the main distribution.

If you want to use it, you'll have to use this PR: https://github.com/ros-industrial/ur_modern_driver/pull/216.

gavanderhoorn, 2019-06-20:
As this is not an issue with the packages in this repository, I'm going to close this.

Feel free to keep commenting on it of course, but please be aware of the fact that discussing anything related to e-series and drivers should not happen here, but on the `ur_modern_driver` tracker.

closed 2019-06-20 by gavanderhoorn

Drikuy, 2019-06-20:
Thanks for your help!