ros-industrial/universal_robot #440: Suggested new repository structure

gavanderhoorn, 2019-07-02:
During WRID19 we (RIC-EU) have discussed the future of the packages in this repository and recongised the following challenges/constraints/requirements:

 1. the description, moveit and kinematics package(s) should probably stay together: changes to one often automatically need changes in the other(s), so having their release cycle synchronised makes sense
 1. `ur_driver` is deprecated and should not be used any longer
 1. `ur_msgs` is currently used by `ur_modern_driver` *and* `ur_driver`, but is not necessarily tied to the state of any of the other packages. `ur_driver` is going to be retired, which leaves just `ur_modern_driver`. However, the new UR driver (FZI & UR) will also most likely also use `ur_msgs`
 1. `ur_gazebo` currently is almost empty (some `ros_control` config and `.launch` files), with most of the Gazebo-specific content actually hosted in `ur_description`. 

This makes `ur_gazebo` relatively stand-alone, with no apparent reason to be hosted in `ros-industrial/universal_robot`. However, gazebo is "just one" simulator (although a popular one in ROS), and there is actually no need to directly include

gavanderhoorn, 2019-07-02:
@ipa-nhg: FYI.

@miguelprada also.

miguelprada, 2019-07-02:
I see one issue with the suggested layout.

We've been recently receiving a lot of issues by people getting confused about how to "connect" the MoveIt setup to the actual robot driver. With the proposed layout, this is very likely to continue being the case, since the correct MoveIt setup will depend on the topic _layout_ exposed by each specific driver.

Not sure what the best way to avoid this is. One option would be to include MoveIt-specific launch files to the driver repositories `3.`/`4.`. Other option would be by writing clear documentation on how to adapt the existing MoveIt configuration in `1.`, and making it crystal clear that in its default state it's only supposed to work in simulation (may also work with some specific driver setup, but it's better to assume it won't). 

I don't quite like any of these options. More ideas?

gavanderhoorn, 2019-07-02:
That is a good point.

Convenience `moveit_planning_execution.launch` files could be an idea, but we could also expect the user to start the driver and MoveIt seperately.

At a minimum what would be needed would be a custom `controllers.yaml` with the correct topic/action names (namespaces, etc). However that doesn't really work very well to instruct users to use those.

I'm not sure right now what would be the most convenient way to do this, without delegating some aspect of this to the user btw.

Ideally all these packages get released, which would make them read-only. Any solution would have to be usable without *everyone* needing to edit the packages just to be able to get started.

gavanderhoorn, 2019-07-02:
@fmauch.

fmauch, 2019-07-02:
We could also include moveit launchfiles into the drivers that utilize `ur_x_moveit_config` and map the correct controller.

gavanderhoorn, 2019-07-02:
> We could also include moveit launchfiles into the drivers that utilize `ur_x_moveit_config` and map the correct controller.

I believe that is what @miguelprada called "moveit convenience launch files".

gavanderhoorn, 2019-07-13:
Just a thought but it would be much simpler to do these kinds of things if it were possible to somehow "overlay" or contribute additional files to already existing packages from other packages.

If that were possible, we could simple have `ur_modern_driver` add `moveit_planning_execution.launch` files to the MoveIt configuration files.

fmauch, 2022-11-08:
I think, this can be closed, since everything from this issue is done.

closed 2022-11-08 by fmauch