ros-industrial/universal_robot #442: `ur_kinematics` issues and future work?

Yuki-cpp, 2019-07-04:
We were looking at `ur_kinematics` and noticed that it seems to have received little attention over the last year despite a number of issues. A project has been created that consolidates all of these issues, but is there a roadmap to address them? If not, we will modify `ur_kinematics` to fit our own purposes, including a fairly large refactoring that would probably break existing API but would make the package more functional. Would you be interested in this sort of work?

gavanderhoorn, 2019-07-04:
Yes, there would certainly be interest.

However, `ur_kinematics` is first and foremost a MoveIt IK plugin for the UR MoveIt packages. That functionality must be preserved in some way (through a shim or some other way).

Second: the currently outstanding PRs seem to resolve quite a few issues and I'd like those to be integrated instead of ignored.

re: planning: properly dealing with IK-related issues is non-trivial and they potentially have a large impact if handled incorrectly. There simply hasn't been someone taking this on. If you'd be willing to do that, I'd fully support it.

haudren, 2019-07-08:
I'm piggybacking on this issue to give a little update:
- We started some work on this, we'll keep you updated
- I noticed that #358 would bring a lot of improvements, any idea what's blocking here? We would like to base our improvements on it.

Thanks,

gavanderhoorn, 2019-07-08:
> We started some work on this, we'll keep you updated as soon as we have something worth showing

may I suggest to provide updates sooner and more often?

It would be very unfortunate if you spend a lot of effort only to not pass a (PR) review later.

> I noticed that #358 would bring a lot of improvements, any idea what's blocking here? We would like to base our improvements on it.

As I wrote in my first response:

> re: planning: properly dealing with IK-related issues is non-trivial and they potentially have a large impact if handled incorrectly. **There simply hasn't been someone taking this on.**

It's a matter of resources.

gavanderhoorn, 2019-07-13:
@Yuki-cpp @haudren: have you made any progress?

gavanderhoorn, 2019-09-02:
Ping @Yuki-cpp and @haudren.