ros-industrial/universal_robot #444: melodic-devel behind kinetic-devel?

v4hn, 2019-07-05:
> melodic-devel is 7 commits behind kinetic-devel. 

Could you please fast-forward melodic-devel?

gavanderhoorn, 2019-07-05:
No.

`melodic-devel` is currently being used for some specific purposes.

It's at a specific commit for a reason.

I suggest you use `kinetic-devel` for now.

v4hn, 2019-07-05:
That statement sounds rather mysterious.

If the latest changes in kinetic-devel are not relevant for melodic, there is indeed no reason to change anything, but it did not seem that way to me.

closed 2019-07-05 by v4hn

gavanderhoorn, 2019-07-05:
> If the latest changes in kinetic-devel are not relevant for melodic

That's not what I'm saying.

What I meant was that `kinetic-devel` is the branch to use for both Kinetic and Melodic. `melodic-devel` will become the branch to use in the near future (probably even also for Kinetic), but not right now.

There are no changes in `melodic-devel` right now compared to `kinetic-devel` that would make you want to use `melodic-devel`, other than perhaps the fact that the name of the branch corresponds with the name of the ROS release you're using.

v4hn, 2019-07-05:
> other than perhaps the fact that the name of the branch corresponds with the name of the ROS release you're using.

... which feels like a good-enough reason to me ...

Thanks for elaborating!

davetcoleman, 2020-01-21:
Moving my discussion from https://github.com/ros-industrial/universal_robot/issues/475 here:

> What I meant was that kinetic-devel is the branch to use for both Kinetic and Melodic. melodic-devel will become the branch to use in the near future (probably even also for Kinetic), but not right now.

This is very confusing for most ROS developers who want to contribute open source. @gavanderhoorn you're making this very hard for me (and anyone else) to improve the very popular UR drivers. Do you not want contributors? I can just make yet another fork and have my own melodic drivers, but I don't want to do that.

I need to make melodic-only changes to the UR driver, due to melodic-only changes in MoveIt.

Your stance on branching of this repo is very confusing to most anyone but you. Please consider deleting the melodic-devel branch until its mysteriously ready to be used.

Also please consider adding clarification of this stance in the README