ros-industrial/universal_robot #446: Action client not connected in ROS Melodic

vkguda355, 2019-07-11:
Hi,

I have the melodic version of the ur5 package 

I can properly connect to the robot typing:
roslaunch ur_moder_drivers ur5_bringup.launch robot_ip:=IP_OF_THE_ROBOT 

however, when I try to run the moveit planner typing:
roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch 

I get the error messange:

[ERROR] [....]: MoveitSimpleControllerManager: Action client not connected: /follow_joint_trajectory

How can I fix this issue?
This issue, by the way, it was working few hours back without any problem.

I followed the tutorial : http://wiki.ros.org/universal_robot/Tutorials/Getting%20Started%20with%20a%20Universal%20Robot%20and%20ROS-Industrial 

thanks for helping

gavanderhoorn, 2019-07-12:
> it was working few hours back without any problem.

then I would suggest to first try and recreate the situation in which it was working.

vkguda355, 2019-07-12:
No , I tried multiple times to check I still he the same error message. 0 controlloin the list and action client not connected

vkguda355, 2019-07-12:
When I run the modern drivers, and check for the rostopics, I don't find the / follow_joint _rajectory topics listed

gavanderhoorn, 2019-07-12:
Then it would seem the driver is not correctly configured or even running.

Can you show all console output that results from running `roslaunch ur_moder_drivers ur5_bringup.launch robot_ip:=IP_OF_THE_ROBOT`?

vkguda355, 2019-07-12:
yes i am using the ur_modern_drivers, have given the actual ip address.and i dont know which branch i had cloned. it was long time back.

`PC:~/catkin_ws$ roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=192.168.0.11
... logging to /home/vkguda/.ros/log/d68e88c0-a480-11e9-9d48-d8c49799a17d/roslaunch-vkguda-PC-4292.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://vkguda-PC:39179/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /ur_driver/base_frame: base
 * /ur_driver/max_joint_difference: 0.01
 * /ur_driver/max_payload: 5.0
 * /ur_driver/max_velocity: 10.0
 * /ur_driver/max_waiting_time: 2.0
 * /ur_driver/min_payload: 0.0
 * /ur_driver/prefix: 
 * /ur_driver/require_activation: Never
 * /ur_driver/reverse_ip_address: 
 * /ur_driver/reverse_port: 50001
 * /ur_driver/robot_ip_address: 192.168.0.11
 * /ur_driver/servoj_gain: 100.0
 * /ur_driver/servoj_lookahead_time: 1.0
 * /ur_driver/servoj_time: 0.008
 * /ur_driver/servoj_time_waiting: 0.001
 * /ur_driver/shutdown_on_disconnect: True
 * /ur_driver/time_interval: 0.008
 * /ur_driver/tool_frame: tool0_controller
 * /ur_driver/use_lowbandwidth_trajectory_follower: False
 * /ur_driver/use_ros_control: False

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ur_driver (ur_modern_driver/ur_driver)

auto-starting new master
process[master]: started with pid [4307]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d68e88c0-a480-11e9-9d48-d8c49799a17d
process[rosout-1]: started with pid [4318]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [4321]
process[ur_driver-3]: started with pid [4325]
[ INFO] [1562920888.485243293]: Setting up connection: 192.168.0.11:30003
[ INFO] [1562920888.485872819]: Connection established for 192.168.0.11:30003
[ INFO] [1562920888.485899381]: Setting up connection: 192.168.0.11:30001
[ INFO] [1562920888.486333119]: Connection established for 192.168.0.11:30001
[ INFO] [1562920888.486888938]: Got VersionMessage:
[ INFO] [1562920888.486902258]: project name: URControl
[ INFO] [1562920888.486909823]: version: 3.6.0
[ INFO] [1562920888.486916428]: build date: 14-05-2018, 16:43:28
[ INFO] [1562920888.486923168]: Disconnecting from 192.168.0.11:30001
[ INFO] [1562920888.491141697]: ActionServer enabled
[ INFO] [1562920888.491657095]: Use standard trajectory follower
[ INFO] [1562920888.492328532]: Setting up connection: :50001
[ INFO] [1562920888.492421500]: Connection established for :50001
[ INFO] [1562920888.492461823]: Initializing ur_driver/URScript subscriber
[ INFO] [1562920888.493985968]: The ur_driver/URScript initialized
[ INFO] [1562920888.494010425]: Notifier: Pipeline disconnect will shutdown the node
[ INFO] [1562920888.496059803]: Service 'ur_driver/robot_enable' activation mode: Never
[ INFO] [1562920888.496079822]: Starting main loop
[ INFO] [1562920888.496148861]: Setting up connection: 192.168.0.11:30003
[ INFO] [1562920888.496178402]: Starting pipeline RTPacket
[ INFO] [1562920888.496233940]: Starting pipeline StatePacket
[ INFO] [1562920888.496271452]: Setting up connection: 192.168.0.11:30002
[ INFO] [1562920888.496582722]: Connection established for 192.168.0.11:30003
[ INFO] [1562920888.496623195]: Connection established for 192.168.0.11:30002
[ INFO] [1562920888.496904716]: Starting ActionServer
[ INFO] [1562920888.496974241]: Trajectory thread started

`

gavanderhoorn, 2019-07-12:
First: please make sure you're using the latest version of `ur_modern_driver` (`git pull` in the `ur_modern_driver` package directory).

Start the driver again, and then show us the output of `rostopic list`.