ros-industrial/universal_robot #447: UR urdf

ghx-0228, 2019-07-12:
Hey, I am just curious about a problem with the UR simulation in Gazebo.

I find all links connected to "wrist_3_link" can not display in the structure tree at left panel, like "ee_link" or "tool0".  But  they indeed are in the model, after adding visual property for them in urdf, we can see them emerge in Scene zone. I infer and find they all become the nested model of "wrist_3_link".   However, I check the urdf file and did not find what makes these model become nested model. So could you help me or tell me how to implement this in urdf.
[image]

gavanderhoorn, 2019-07-12:
Does your model look like the one shown in #400?

If so: check whether #374 describes your situation and the solution.

gavanderhoorn, 2019-07-12:
I'm going to close this for now as I believe this is a user-side configuration issue (caused by ros/urdfdom_headers#41).

If it turns out that is not the cause, we can re-open.

closed 2019-07-12 by gavanderhoorn

ghx-0228, 2019-07-12:
Thanks for your reply.
I have also tried to add visual and collision properties for ee_link. But the result is the same. We can see there is a grey box on the UR but can not see in the structure tree. 
[image]
Additionally, if I connect a part to "wrist_3_link", it can be one part with "wrist_3_link" instead of  a separate part.
[image]

gavanderhoorn, 2019-07-12:
Ok, I believe I know what is going on.

First a general comment:

> I have also tried to add visual and collision properties for ee_link. 

`ee_link` and `tool0` are not supposed to be edited. You should not add `visual` nor `collision` geometry to them, nor should you edit their location.

`ee_link` is deprecated and will go away. A `flange` frame will be added instead.

> But the result is the same. We can see there is a grey box on the UR but can not see in the structure tree.

This is a property of Gazebo: when loading a `.urdf`, Gazebo will merge all links that are only connected via `fixed` joints.

As both `ee_link` and `tool0` are connected via a `fixed` joint to `wrist_3_link`, you will not find `ee_link` nor `tool0` in the model tree of Gazebo, as they are pruned from the model.

This is something that Gazebo does. We can't influence that.

ghx-0228, 2019-07-12:
OK, get it. Thanks very much!