ros-industrial/universal_robot #448: Breaking changes in Melodic release

gavanderhoorn, 2019-07-13:
This is a meta-issue / discussion issue for the first Melodic release, which is planned to include a number of breaking changes.

Summarising for now:

 - `universal_robot` (metapackage):
   - will be removed (`universal_robots` replaces it, `universal_robot` depends on `universal_robots` right now)

 - `ur_driver` (#440)
   - will be removed: `ur_modern_driver` will become the standard driver for CB1 and CB2 controllers

 - `ur_bringup` (#429, #440)
   - will be removed (instead: drivers should provide the necessary `.launch` files)

 - MoveIt configuration packages:
   - updated to include latest MoveIt functionality (#430)
   - files and layout will follow ROS-I standards (`moveit_planning_execution.launch`, etc) (#431)

 - `ur_description`:
   - merged with `ur_e_description`
   - support for calibration data imported from robot controller (#414, #357)
   - "joint limited" variants will be removed
   - migration to ROS-I standard structure with top-level and `_macro.xacro` and standard `.launch` files (started with #435)
   - heavy use of `xacro` yaml support (#371)
   - single pair of top-level and `_macro.xacro` which will be parameterised with `.yaml` settings files: one for each UR model (both e-series and regular robots) (#371)
   - removal of the `world` link (#284 and https://github.com/ros-industrial/universal_robot/pull/371#issuecomment-510568694)
   - removal of all Gazebo-specific infrastructure and settings (will be migrated to `ur_gazebo`)

 - `ur_gazebo`:
   - will use similar `.yaml` based approach as `ur_description`
   - will provide a `_macro.xacro` that adds Gazebo-specific elements and plugins (similar to (but not necessarily identical to) [ros-industrial/abb_experimental/abb_irb120_gazebo](https://github.com/ros-industrial/abb_experimental/tree/35af6b1a359777052292dfae834535d79bc4d725/abb_irb120_gazebo))
   - will provide `.launch` files to load `robot_description` and other parameters (`gazebo_ros_control` configuration fi)
   - migration to `ros_control` effort controllers for all simulated robots

 - `ur_e_gazebo`:
   - will be merged with `ur_gazebo`

 - `ur_msgs`:
   - will be moved to a separate repository (#440)

gavanderhoorn, 2019-07-13:
@ipa-nhg @ipa-led @miguelprada

fmauch, 2020-06-15:
As I couldn't find it anywhere: It might be worthwhile mentioning explicitly that with #371 the `urX_e_moveit_config` packages are broken, as they refer to `ur_e_description` and `urdf/urX_e_robot.urdf.xacro`.

I assume that this in scope of #431?

gavanderhoorn, 2020-06-15:
They're broken in `melodic-devel-staging`.

That is known and will be fixed.

I decided not to fix them by not including #467 in #501.

fmauch, 2020-06-16:
I could maybe start hacking on this moveit stuff. I'm not very familiar with moveit, so it's probably going to take me a while... But I'd like pushing this forward :-)

gavanderhoorn, 2020-06-16:
That would be appreciated.

I have some pretty specific ideas about those configuration pkgs though, so it may take a bit of back-and-forth.

The MSA v2 does some strange things I don't want to expose new users to, and we abstract away the differences between using MoveIt with a real controller and a Gazebo robot (the required remap fi will go away), so I don't like any mention of Gazebo in the packages. The new MSA adds some files for Gazebo support, so those would need to be removed.

fmauch, 2020-06-16:
I'll try to get familiar with moveit first and then create a Draft-PR outlining the changes I would do. Then we can discuss that there before I dig down implementing too deep.

fmessmer, 2020-11-24:
I guess this is still the best issue to ask for the progress of `melodic-devel-staging`
I'm having a hard time to puzzle together compatible set of `universal_robot`, `Universal_Robots_ROS_Driver` and `ur_msgs` branches providing all the latest features - including support for the e-series

Any suggestions what the best combo should be?
Any plans on finally getting things streamlined into each repositories' default branches again?

gavanderhoorn, 2020-11-24:
I just followed the instructions [in the readme](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/777560cf735d33c706a1e5cfc6a8daeffc6a0b54/README.md#building) of `Universal_Robots_ROS_Driver` in a clean Docker container (copy-pasting every line) which resulted in a successful build.

Does that not work for you?

fmessmer, 2020-11-24:
well, the readme still revers to using `fmach/universal_robot@calibration-devel` which is not really up to date with `melodic-devel-staging`
I thought those two repos (`universal_robot`, `Universal_Robots_ROS_Driver`) go in sync...

gavanderhoorn, 2020-11-24:
No, not necessarily.

fmauch, 2020-11-24:
As long as the README of `Universal_Robots_ROS_Driver` suggests to use my fork, that's still the suggested way to go. The two do not get synced regularly, as the approaches for doing the calibration correction are a bit different. Due to that, the version on my fork isn't as "breaking" as this `melodic-devel-staging` and e.g. the existing moveit integrations can be used. From my point of view I would switch the suggestion once #538 is resolved.

v4hn, 2021-05-04:
Ping. (@gavanderhoorn ?)

The discussion on breaking changes in the melodic release has been going on for almost two years, bigger groups migrate to noetic by now and this is still an issue:

```
❯ rosinstall_generator --rosdistro melodic ur_description
The following unreleased packages/stacks will be ignored: ur_description
No packages/stacks left after ignoring unreleased
```

We noticed in the moveit_tutorials because [rosdep can't find the packages when they're not compiled in the same workspace](https://github.com/ros-planning/moveit_tutorials/pull/558#discussion_r625529018). However, they are not actually build_depends, so it's not really justified to build them in CI unless a test would rely on them...

fmauch, 2022-11-10:
Looks like everything here is done -> closing

closed 2022-11-10 by fmauch