ros-industrial/universal_robot #449: End effector wobbling after update.

gstavrinos, 2019-07-15:
Dear ROS-I developers,

We have a UR3+Schunk PG70 ( https://github.com/Roboskel-Manipulation/manos/ )combo that was working flawlessly both in simulation and with real hardware. After commit 62011b0d353841c0c219d862c359c540a4bc6aee, we encountered weird wobbling when the arm was resting, and the gripper's fingers completely snapped when the arm started moving.

I can provide you with a video if needed.

Have you also encountered this weird behaviour after that specific commit?

Thanks in advance,

George

gavanderhoorn, 2019-07-15:
> Have you also encountered this weird behaviour after that specific commit?

Well, no, mostly because we (or I) don't have your specific setup I guess.

> I can provide you with a video if needed.

that would help, yes.

62011b0 fixed the inertia parameters for all robots, and for the UR3 [these lines where changed](https://github.com/ros-industrial/universal_robot/commit/62011b0d353841c0c219d862c359c540a4bc6aee#diff-bc1e6c4a2c3a755e869b2c79c18adeea).

If you revert those, do things work like they used to? More specifically: could you check which of those you'd have to revert to get things to stabilise again?

As the previous settings were just wrong (as explained by @fmauch), it might be that your previous setup being stable was a fluke (ie: stable with the incorrect characteristics of the robot model).

Note that we're happy to assist, but we can't help getting (Gazebo) simulations stable, as it's just too finicky a business, and typically very brittle as well.

gstavrinos, 2019-07-15:
Thanks for the quick response. It is completely understandable to not being able to support gazebo-related issues.

I have cornered the problem to the `length`="0.04" part of the `cylinder_inertial` parameter of the `wrist_3_link`. When I revert it back to 0.12, everything works as expected.

Here is a quick demonstration of the problem:

[image]

gavanderhoorn, 2019-07-15:
@fmauch: would you happen to have any idea?

fmauch, 2019-07-15:
I'll have to have a look at it. As far as I understand it, you are simply starting the launchfile from the linked package? That should be enough to reproduce the problem.

gstavrinos, 2019-07-15:
Yes, I am just running this command:
`roslaunch manos_gazebo manos_gazebo.launch`

Just an extra note: I also have to revert the `roll ` coordinate of `wrist_3_link` back to 0, in addition to the length fix, in order for the robot to stop wobbling.

fmauch, 2019-07-22:
I spent quite some time today to further investigate this. Unfortunately I did not find a solution for this.

I agree with @gavanderhoorn that the working state from before this commit was most likely not the really "correct" state. But I agree that the dimensions entered in my commit seem not to be correct either.

I'll need some more time / knowledge to further investigate this.

fmauch, 2019-07-22:
I reshaped the inertia approximation by using the mesh bounding boxes. (For the UR3 those were indeed quite large).

@gstavrinos I pushed my changes to https://github.com/fmauch/universal_robot/tree/fix_ur3

It seems as the wobbling in the starting state seems to be resolved by that. Could you please try moving it around a bit?

gstavrinos, 2019-07-23:
Thanks @fmauch for taking the time to investigate this. Although the robot stopped wobbling when idle, unfortunately it still freaks out when moving.

gavanderhoorn, 2020-06-11:
It could be that #504 will help with this.

But seeing as it has been a long time since there was any activity here, I'm going to close it.

If this is still an issue @gstavrinos, please comment here and we can re-open.

closed 2020-06-11 by gavanderhoorn