ros-industrial/universal_robot #451: issues occured using trac_ik inverse solver

TheJasal, 2019-07-19:
I run roslaunch ur5_moveit_config demo.launch,and got this trac_ik plugin error as bellow:

[ERROR] [1563527196.206434504]: The kinematics plugin (manipulator) failed to load. Error: According to the loaded plugin descriptions the class trac_ic_kinematics_plugin/TRAC_IKKinematicsPlugin with base class type kinematics::KinematicsBase does not exist. Declared types are  cached_ik_kinematics_plugin/CachedKDLKinematicsPlugin cached_ik_kinematics_plugin/CachedSrvKinematicsPlugin kdl_kinematics_plugin/KDLKinematicsPlugin lma_kinematics_plugin/LMAKinematicsPlugin pr2_arm_kinematics/PR2ArmKinematicsPlugin srv_kinematics_plugin/SrvKinematicsPlugin trac_ik_kinematics_plugin/TRAC_IKKinematicsPlugin ur_kinematics/UR10KinematicsPlugin ur_kinematics/UR3KinematicsPlugin ur_kinematics/UR5KinematicsPlugin
[ERROR] [1563527196.206460852]: Kinematics solver could not be instantiated for joint group manipulator.

The libtrac_ik_kinematics_plugin.so file does exit in /opt/ros/kinetic/lib and /catkin_ws/src/universal_robot/ur_kinematics.One strange thing is that when I use other kinematics plugins instead of trac_ik,for example kdl,it won't throw out this error.So I run ~ sudo apt-get install ros-kinetics-trac_ik,but it does not work.

So anybody has this problem?please tell me how to fix it,thanks a lot!

gavanderhoorn, 2019-07-19:
Without diagnosing this further I wouldn't know what is causing this.

Some general troubleshooting tips:

 - make sure you have sourced the correct workspace(s)
 - make sure the package is installed
 - make sure there are no typos in the plugin name
 - run `rospack profile`

gavanderhoorn, 2019-07-19:
As we don't ship any moveit configurations which use `trac_ik`, this is not a problem with any of the packages in this repository.

As such, I'm closing this issue.

Feel free to keep commenting on it, but please understand we cannot provide support for all combinations of packages.

closed 2019-07-19 by gavanderhoorn