ros-industrial/universal_robot #452: ERROR: cannot launch node of type [moveit_ros_move_group/move_group]: can't locate node [move_group] in package [moveit_ros_move_group]

anubhav1772, 2019-08-18:
I am getting above error when trying to run a launch file in ros. 

Below is the screenshot of the error:
[image]

Result of `env|grep ROS` command:
```
ROS_ETC_DIR=/opt/ros/melodic/etc/ros
ROS_ROOT=/opt/ros/melodic/share/ros
ROS_MASTER_URI=http://localhost:11311
ROS_VERSION=1
ROS_PYTHON_VERSION=2
ROS_PACKAGE_PATH=/home/anubhav1772/catkin_ws:/opt/ros/melodic/share
ROSLISP_PACKAGE_DIRECTORIES=
ROS_DISTRO=melodic
```
And, below is the result of `rospack find moveit_ros_move_group` command:

`/opt/ros/melodic/share/moveit_ros_move_group`

So, from above it seems that `moveit_ros_move_group` package is already installed, but I don't know why it's not able to locate the package. 

Below is the part of `move_group.launch` file where I am trying to use above package.

```
<!-- Start the actual move_group node/action server -->
<node name="move_group" launch-prefix="$(arg launch_prefix)" pkg="moveit_ros_move_group" type="move_group" respawn="false" output="screen" args="$(arg command_args)">
 <!-- Set the display variable, in case OpenGL code is used internally -->
 <env name="DISPLAY" value="$(optenv DISPLAY :0)" />

 <param name="allow_trajectory_execution" value="$(arg allow_trajectory_execution)"/>
 <param name="max_safe_path_cost" value="$(arg max_safe_path_cost)"/>
 <param name="jiggle_fraction" value="$(arg jiggle_fraction)" />
 <param name="capabilities" value="$(arg capabilities)"/>
 <param name="disable_capabilities" value="$(arg disable_capabilities)"/>

 <!-- Publish the planning scene of the physical robot so that rviz plugin can know actual robot -->
 <param name="planning_scene_monitor/publish_planning_scene" value="$(arg publish_monitored_planning_scene)" />
 <param name="planning_scene_monitor/publish_geometry_updates" value="$(arg publish_monitored_planning_scene)" />
 <param name="planning_scene_monitor/publish_state_updates" value="$(arg publish_monitored_planning_scene)" />
 <param name="planning_scene_monitor/publish_transforms_updates" value="$(arg publish_monitored_planning_scene)" />
</node>
```
You help would be highly appreciated. Thanks.

gavanderhoorn, 2019-08-19:
I'm slightly unsure as to why you post your issue here, on this issue tracker: it doesn't appear to have any relation to the packages in this repository. This is not a generic support channel or forum.

As such I'm closing this.

If you feel this is related to `universal_robot` packages, then please clarify, and we can see whether it makes sense to re-open.

In any case feel free to keep commenting on the issue of course.

closed 2019-08-19 by gavanderhoorn

anubhav1772, 2019-08-19:
Sorry for raising above issue at wrong place. I successfully solved the above problem.

Link: https://github.com/ros-planning/moveit/issues/1632