ros-industrial/universal_robot #458: when i run catkin_make, the problem occured. ImportError: No module named 'error'

Liu-Xv, 2019-10-02:
[ 89%] Built target ur3_moveit_plugin
[ 92%] Built target ur10_moveit_plugin
[ 95%] Built target ur5_moveit_plugin
[ 96%] Generating Python code from SRV ur_msgs/SetSpeedSliderFraction
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/genpy/cmake/../../../lib/genpy/gensrv_py.py", line 43, in <module>
    import genpy.generator
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/__init__.py", line 34, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 44, in <module>
    import yaml
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 2, in <module>
    from error import *
ImportError: No module named 'error'
Traceback (most recent call last):
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/genpy/cmake/../../../lib/genpy/genmsg_py.py", line 44, in <module>
  File "/opt/ros/kinetic/share/genpy/cmake/../../../lib/genpy/genmsg_py.py", line 44, in <module>
    import genpy.generator
    import genpy.generator
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/__init__.py", line 34, in <module>
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/__init__.py", line 34, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 44, in <module>
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 44, in <module>
    import yaml
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 2, in <module>
    import yaml
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 2, in <module>
    from error import *
ImportError: No module named 'error'
    from error import *
ImportError: No module named 'error'
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/genpy/cmake/../../../lib/genpy/genmsg_py.py", line 44, in <module>
    import genpy.generator
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/__init__.py", line 34, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 44, in <module>
    import yaml
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 2, in <module>
    from error import *
ImportError: No module named 'error'
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/genpy/cmake/../../../lib/genpy/genmsg_py.py", line 44, in <module>
    import genpy.generator
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/__init__.py", line 34, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 44, in <module>
    import yaml
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 2, in <module>
    from error import *
ImportError: No module named 'error'
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/genpy/cmake/../../../lib/genpy/genmsg_py.py", line 44, in <module>
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/genpy/cmake/../../../lib/genpy/genmsg_py.py", line 44, in <module>
    import genpy.generator
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/__init__.py", line 34, in <module>
    import genpy.generator
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/__init__.py", line 34, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 44, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 44, in <module>
    import yaml
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 2, in <module>
    import yaml
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 2, in <module>
[... 35 lines cut ...]
universal_robot/ur_msgs/CMakeFiles/ur_msgs_generate_messages_py.dir/build.make:99: recipe for target '/home/liuxu/catkin_ws/devel/lib/python3/dist-packages/ur_msgs/msg/_MasterboardDataMsg.py' failed
make[2]: *** [/home/liuxu/catkin_ws/devel/lib/python3/dist-packages/ur_msgs/msg/_MasterboardDataMsg.py] Error 1
universal_robot/ur_msgs/CMakeFiles/ur_msgs_generate_messages_py.dir/build.make:84: recipe for target '/home/liuxu/catkin_ws/devel/lib/python3/dist-packages/ur_msgs/msg/_Digital.py' failed
make[2]: *** [/home/liuxu/catkin_ws/devel/lib/python3/dist-packages/ur_msgs/msg/_Digital.py] Error 1
universal_robot/ur_msgs/CMakeFiles/ur_msgs_generate_messages_py.dir/build.make:114: recipe for target '/home/liuxu/catkin_ws/devel/lib/python3/dist-packages/ur_msgs/srv/_SetIO.py' failed
make[2]: *** [/home/liuxu/catkin_ws/devel/lib/python3/dist-packages/ur_msgs/srv/_SetIO.py] Error 1
universal_robot/ur_msgs/CMakeFiles/ur_msgs_generate_messages_py.dir/build.make:79: recipe for target '/home/liuxu/catkin_ws/devel/lib/python3/dist-packages/ur_msgs/msg/_RobotModeDataMsg.py' failed
make[2]: *** [/home/liuxu/catkin_ws/devel/lib/python3/dist-packages/ur_msgs/msg/_RobotModeDataMsg.py] Error 1
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/genpy/cmake/../../../lib/genpy/gensrv_py.py", line 43, in <module>
    import genpy.generator
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/__init__.py", line 34, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 44, in <module>
    import yaml
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 2, in <module>
    from error import *
ImportError: No module named 'error'
universal_robot/ur_msgs/CMakeFiles/ur_msgs_generate_messages_py.dir/build.make:119: recipe for target '/home/liuxu/catkin_ws/devel/lib/python3/dist-packages/ur_msgs/srv/_SetSpeedSliderFraction.py' failed
make[2]: *** [/home/liuxu/catkin_ws/devel/lib/python3/dist-packages/ur_msgs/srv/_SetSpeedSliderFraction.py] Error 1
CMakeFiles/Makefile2:962: recipe for target 'universal_robot/ur_msgs/CMakeFiles/ur_msgs_generate_messages_py.dir/all' failed
make[1]: *** [universal_robot/ur_msgs/CMakeFiles/ur_msgs_generate_messages_py.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed

gavanderhoorn, 2019-10-02:
> `/home/liuxu/catkin_ws/devel/lib/python3/dist-packages`

are you creating a mixed Python 2 & Python 3 workspace?

The error you show seems to be a generic Python error, not something specific to this repository or caused by any particular package here.

Can you build any messages in a ROS workspace?

Liu-Xv, 2019-10-02:
Thank you for your help! I have solved the problem. It seems like I wrongly run the command "catkin_make -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.7m -DPYTHON_LIBRARY=/usr/lib/libpython3.7m.so". I delete this folder and mkdir a new one, then the problem disappered.

closed 2019-10-02 by gavanderhoorn