ros-industrial/universal_robot #460: Add support for UR16e

gavanderhoorn, 2019-10-11:
Recently released.

`ur_support` should be extended to include meshes, config, launch files and xacro macro.

fmauch, 2020-01-07:
I started working on that on a train trip during the Christmas holidays. For the sake of simplicity I decided to start working on top of ros-industrial/universal_robot#371. The current status is on https://github.com/fmauch/universal_robot/tree/ur16e

Things still to be done:
 - [x] Add config files for ur16e
 - [x] Add description launch files for ur16e
 - [x] Add gazebo launchfile for ur16e
 - [ ] Add meshes of upper and lower arm. To my knowledge all other segments are the same as for the ur10e. The arm segments are also the same, but shortened. So, taking out a part in the middle should be sufficient **Help needed here**
 - [ ] Update physical and visual parameters. I started working on those, but many of them are copied from the ur10e
 - [ ] Add `ur_kinematics` files for the ur16e
 - [ ] Add `ur16_e_moveit_config`

I hope, this covers everything otherwise please add more points, I'll add them to the list.

arthurpetron, 2020-01-11:
I have urdfs + unstyled meshes for the UR16e if you'd like me to send them to you. Have not broken the model up into separate visual and collision meshes yet, but it's a start.

gavanderhoorn, 2020-01-11:
We could certainly take a look at them @arthurpetron.

Would you have them somewhere available in a repository? That would be most convenient.

arthurpetron, 2020-01-24:
Sure, try here: https://github.com/arthurpetron/ur16e-dev . The inertial values are a rough estimate, but the masses should be correct.

fmauch, 2020-01-28:
@arthurpetron Some parameters (such as mass and DH-converted kinematics) from [eSeries_UR16e_15012020_URDF_ONLY.csv](https://github.com/arthurpetron/ur16e-dev/blob/master/eSeries_UR16e_URDF/urdf/eSeries_UR16e_15012020_URDF_ONLY.csv) differ from https://www.universal-robots.com/how-tos-and-faqs/faq/ur-faq/parameters-for-calculations-of-kinematics-and-dynamics-45257/.

Are they extracted based on the solidworks model?

Apart from that I took all of my rudimentary blender skills and chopped out a piece from the ur10e upperarm and forearm models to get the correct meshes.

arthurpetron, 2020-01-28:
The mass values are all correct (entered by hand) the intertial values are not. They are extracted from the solidworks URDF creation. UR does provide the full, 3x3 intertial matrices. I've copied them here: `inertia_matrix = [[0.03351, 0.00002, -0.00001, 0.00002, 0.03374, 0.00374, -0.00001, 0.00374, 0.02100], [0.02796, -0.00010, -0.00720, -0.00010, 0.47558, 0.00003, -0.00720, -0.00001, 0.47635], [0.01091, 0.00006, 0.01012, 0.00006, 0.12060, 0.00001, 0.01012, 0.00001, 0.11714], [0.00609, -0.00001, -0.00000, -0.00001, 0.00245, 0.00083, 0.00000, 0.00083, 0.00579], [0.00389, -0.00001, 0.00000, -0.00001, 0.00219, -0.00045, 0.00000, -0.00045, 0.00363], [0.00117, 0.00000,  0.00000, 0.00000, 0.00118, 0.00000, 0.00000, 0.00000, 0.00084] ]`

It's important that the arm is in the position that I've specified in the URDF so that the zero positions line up.

gavanderhoorn, 2020-06-09:
Initial support for UR16e was merged into `melodic-devel-staging`.

We'd still need to check the inertias and other dynamic parameters though, just to make sure they are correct.

@fmauch: which values did you end up adding to your PR?

fmauch, 2020-06-09:
I modeled them similar to the other models already inside the description packages.

gavanderhoorn, 2020-06-09:
And the changes in https://github.com/ipa-led/universal_robot/compare/ur_description_args...fmauch:inertia_from_ur? Especially the later commits seem relevant perhaps?

fmauch, 2020-06-10:
I'll try to come back on that during the weekend

fmauch, 2020-06-11:
I rebased my work and further elaborated it inside #504, as this doesn't only affect the ur16e.

johannesWen, 2021-02-27:
Hello guys, what is the current status of this topic? We would need a MoveIt support for the UR16e. Is there a chance to implement this in the near future? BR

gavanderhoorn, 2021-02-27:
Creating a MoveIt configuration package is just a matter of completing the wizard. For any use other than a quick test you need to do this in any case, so I don't believe you're held back by a MoveIt configuration not being available yet.

fmauch, 2022-07-21:
MoveIt config has been added in #538, updating my [summary comment above](https://github.com/ros-industrial/universal_robot/issues/460#issuecomment-571532451).

fmauch, 2022-11-10:
kinematics has been added, so everything is done. Closing this issue.

PR #616 (2022-11-08): Update ur kinematics
In #613 I collected a couple of reasons why I think that ur_kinematics is currently flawed. This PR addresses a couple of them.

* Adds parameters for all robot models
* Removes the `kinematics_solver_attempts` as suggested in #474
* Corrects plugin name in `kinematics.yaml` files

Since I still see issues with using this plugin due to its problems regarding robot calibration I added a README to inform users about this concern.

This closes #460
closed 2022-11-10 by fmauch