ros-industrial/universal_robot #461: Unable to use ur_kin_py in Python 3.6

hemanthsarabu, 2019-10-22:
I am trying to build and run ur_kin_py for a python 3 catkin workspace but have not been successful so far. I followed instructions from [here](https://github.com/ros-industrial/universal_robot/pull/364#issuecomment-472294282), and replaced the line in the CMake file from `find_package(PythonLibs 2.7 REQUIRED)`
 to `find_package(PythonLibs 3.6 REQUIRED)` and ran 
`
 catkin_make -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so `.

When I then try to run` test_analytical_ik.py` , I see the following error:
``` 
Segmentation fault (core dumped)

```
And when I try to run `import ur_kinematics.ur_kin_py.inverse` in a python session, I see `ImportError: /usr/lib/x86_64-linux-gnu/libboost_python-py27.so.1.65.1: undefined symbol: PyClass_Type
`. I suspect I am building the library against boost for python 2 and not 3 and I'm not sure how to fix that. Has anyone dealt with this before?

Any help would be greatly appreciated!

Thanks!

chinakwy, 2020-07-05:
I'm stuck at this problem :( , did you fix it?
hopefully can get some help
Thanks!

tungkw, 2021-03-15:
when I met the first problem ```Segmentation fault (core dumped)```, what I do is changing the path related to 'pyconfig.h'. In my case is from "/usr/include/python2.7/" to "/usr/include/python3.6/".

For the second problem, I think it would be good to try somethings like ```-lboost_python3 -lboost_numpy3``` when linking in cmake file.

dhled, 2021-09-29:
Please assign this to me :)

gavanderhoorn, 2021-09-29:
While you're at it, you may want to take a look at https://github.com/ros-industrial/universal_robot/projects/1 as well.

There may be PRs there which affect this issue.

dhled, 2021-09-29:
So building using pybind as suggested in #453 seems to solve all python3 problems. I have tested on Python3.6 and 3.8, it seems to work fine.

The only problem is to fix the test_analytical_ik.py for python3 compatibility.

So how we should proceed ? I can try to fix/rebase the PR #453 and python3 compitibility but this might break other IK related PR.

dhled, 2021-09-29:
This one #459 and maybe #358 (which combine 3 PR)

gavanderhoorn, 2021-09-29:
What about getting those merged first/

Would that be an option?

dhled, 2021-09-29:
Yes for #358.

For #459 it's pretty much useless if we merge #453 (or the rebased / updated version)

gavanderhoorn, 2021-09-29:
I'd be ok with merging #358 into `melodic-devel-staging`.

The rest of the IK related PRs should then also go into that branch.

Could you 'review' #358 and post a thumbs up there?

(you did test it, didn't you ? ;) )