ros-industrial/universal_robot #462: ROS Velocity Controller Usage on UR5e

Abbyls, 2019-10-25:
Hi, 

I am using Ubuntu 16.04+ROS Kinetic+Gaebo9.0. I am trying to use 

> velocity_controllers/JointGroupVelocityController

to control joints velocity, but got strange behavior. I have searched for a lot of related issues and still can not find a solution. 

1. I set the controller.yaml in ur_e_gazebo:

> gazebo_ros_control:
   pid_gains:
    shoulder_pan_joint: {p: 100.0, i: 0.0, d: 0.0}
    shoulder_lift_joint: {p: 100.0, i: 0.0, d: 0.0}
    elbow_joint: {p: 100.0, i: 0.0, d: 0.0}
    wrist_1_joint: {p: 100.0, i: 0.0, d: 0.0}
    wrist_2_joint: {p: 100.0, i: 0.0, d: 0.0}
    wrist_3_joint: {p: 100.0, i: 0.0, d: 0.0}

arm_vel_controller:
  type: velocity_controllers/JointGroupVelocityController
  joints:
     - shoulder_pan_joint 
     - shoulder_lift_joint
     - elbow_joint
     - wrist_1_joint
     - wrist_2_joint
     - wrist_3_joint
......

then, run:

> roslaunch ur_e_gazebo ur5e.launch 

What I got in the gazebo:
[image]

From this .gif we can see that all the joints move fast like there is no controller.

2. I comment the gazebo_ros_control in the controller.yaml file, then I got error in the terminal:

> [ERROR] [1572021488.954333021]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/shoulder_pan_joint
[ERROR] [1572021488.955274188]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/shoulder_lift_joint
[ERROR] [1572021488.956086878]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/elbow_joint
[ERROR] [1572021488.956901888]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_1_joint
[ERROR] [1572021488.957837780]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_2_joint
[ERROR] [1572021488.958694529]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_3_joint

and the joint "shoulder_lift_joint" seems can not hold the gravity to drift down to the ground until reach the joint limitation. While other 5 joints keep do not move. 

[image]

Do you have any ideas about this phenomenon? This problem has confused me for a long time, I will appreciate it very much if someone give me some solutions or ideas!

caiobarrosv, 2020-01-24:
You should implement this [Gazebo plugin](https://github.com/ipa320/cob_gazebo_plugins)

- Configure each joint hardware interface as following:

```
<transmission name="${prefix}shoulder_pan_trans">
      <type>transmission_interface/SimpleTransmission</type>
      <joint name="${prefix}shoulder_pan_joint">
        <hardwareInterface>hardware_interface/PositionJointInterface</hardwareInterface>
        <hardwareInterface>hardware_interface/VelocityJointInterface</hardwareInterface>
      </joint>
      <actuator name="${prefix}shoulder_pan_motor">
        <mechanicalReduction>1</mechanicalReduction>
      </actuator>
    </transmission>
```

- Compile the plugin as described [here](http://gazebosim.org/tutorials?tut=plugins_hello_world&cat=write_plugin#CompilingthePlugin)

- Add the gazebo plugin to your robot URDF file. Don't forget to set the robotNamespace if it exists. 

```
<gazebo>
    <plugin name="gazebo_ros_control" filename="libhwi_switch_gazebo_ros_control.so">
      <robotNamespace></robotNamespace>
      <robotSimType>cob_gazebo_ros_control/HWISwitchRobotHWSim</robotSimType>
    </plugin>
  </gazebo>
```

- Load the position and the velocity controle beforehand in your launch file as following:

```
<arg name="controllers" default="pos_based_pos_traj_controller"/>

  <!-- Load and run pos_based_pos_traj_controller -->
  <node name="ros_control_controller_spawner" pkg="controller_manager" type="spawner" respawn="false"
    output="screen" args="$(arg controllers)" />

  <arg name="stopped_controllers" default="joint_group_vel_controller"/>

  <!-- Load this controller but not run at first-->
  <node name="ros_control_controller_manager" pkg="controller_manager" type="controller_manager" respawn="false"
    output="screen" args="load $(arg stopped_controllers)" />
```

- In your code, switch the controller when you want (Tip: setting the velocity to 0 will not stop the robot, for that you will need to set the robot position using position controller)
This command will turn on position controller
```
rosservice.call_service('/controller_manager/switch_controller', [['pos_based_pos_traj_controller'], ['joint_group_vel_controller'], 1])
```
This command will turn on velocity controller
```
rosservice.call_service('/controller_manager/switch_controller', [['joint_group_vel_controller'], ['pos_based_pos_traj_controller'], 1])
```

If you need to check any code feel free to see [this](https://github.com/lar-deeufba/potential_fields_with_aruco). It is far from an optimized ROS code but we are improving it little by little.

Abbyls, 2020-02-20:
Hi @caiobarrosv, sorry for the late reply as I just came back from vacation.  
I have tried your solution and it has solved my problem! Thank you very much!

closed 2020-02-20 by Abbyls

swagatk, 2020-10-05:
I am trying to implement velocity controller for UR5 Gazebo simulator. I followed your steps and built the cob_gazebo_ros_controller plugin as you suggested. I am still getting this error
```
[ERROR] [1601901104.119941722, 324.684000000]: Could not load controller 'joint_group_vel_controller' because controller type 'velocity_controllers/JointGroupVelocityController' does not exist.
[ERROR] [1601901104.120348877, 324.685000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
```
my "vel_controller.yaml" file contains the following:

```
# Pass an array of joint velocities directly to the joints
joint_group_vel_controller:
   type: velocity_controllers/JointGroupVelocityController
   joints:
     - shoulder_pan_joint
     - shoulder_lift_joint
     - elbow_joint
     - wrist_1_joint
     - wrist_2_joint
     - wrist_3_joint  
```
My Gazebo launch file "myur5.gazebo.launch" looks something like this 
```
<snip>
  <param name="robot_description" command="$(find xacro)/xacro '$(find myur5_description)/urdf/myur5_2.urdf.xacro' transmission_hw_interface:=$(arg transmission_hw_interface)" />

  <!-- push robot_description to factory and spawn robot in gazebo -->
  <node name="spawn_gazebo_model" pkg="gazebo_ros" type="spawn_model" args="-urdf -param robot_description -model robot -z 0.0" respawn="false" output="screen" />

  
  <!-- Another controller -->
  <rosparam file="$(find myur5_velocity_controller)/config/vel_controllers.yaml" command="load" />
  <node name="ros_vel_controller_spawner" pkg="controller_manager" type="controller_manager" args="load joint_group_vel_controller" respawn="false" output="screen"  />
</launch>

```
The URDF file `myur5_2.urdf.xacro` loads the `cob_ros_gazebo_controller` plugin using the following code:

```
  <!-- load gazebo controller plugin -->
  <gazebo>
    <plugin name="gazebo_ros_control" filename="libhwi_switch_gazebo_ros_control.so">
      <robotNamespace></robotNamespace>
      <robotSimType>cob_gazebo_ros_control/HWISwitchRobotHWSim</robotSimType>
    </plugin>
  </gazebo>
```
I downloaded the cob_gazebo_plugin and build it inside my src folder using `catkin_make` command and export the gazebo_plugin path:
```
export GAZEBO_PLUGIN_PATH=/home/swagat/catkin_ws2/devel/lib/:$GAZEBO_PLUGIN_PATH

``` 
The plugin library is located at this path:
```
 $ locate libhwi_switch_gazebo_ros_control.so
/home/swagat/catkin_ws2/devel/lib/libhwi_switch_gazebo_ros_control.so

```
I am implementing this with ROS Melodic on Ubuntu 18.04.  Please let me know if I am missing something here. I am stuck here for quite some time and I will be grateful for any help in this regard. Thanks.

gavanderhoorn, 2020-10-05:
@swagatk: I'm not sure I completely follow: the error message you quote just tells us you don't to have the right `ros_control` controllers installed.

That has nothing to do with `cob_gazebo_ros_controller` afaict.

You probably need to install `ros-melodic-velocity-controllers`.

swagatk, 2020-10-05:
@gavanderhoorn : Thank you very much for your quick reply. 
Yes, indeed your suggestion worked for me.  The error was resolved by installing `ros-melodic-velocity-controllers` package.

Also, `cob_gazebo_ros_controller` is not required. I can use the standard `libgazebo_ros_control.so` plugin.

I can check that my velocity_controller is working by running the following command:
```
$ rosrun controller_manager controller_manager list
'joint_state_controller' - 'hardware_interface::JointStateInterface' ( running )
'arm_vel_controller' - 'hardware_interface::VelocityJointInterface' ( running )
```
Still, my problem is not resolved. As you can see, my robot is hanging down. Do I need to set the PID gain values for this controller? The robot used to stand horizontally straight when I use the default `arm_controll_ur5.yaml` file that comes with the `universal_robot` package. Any suggestions will be of great help. Thanks in advance. 
 
[image]

jaybrecht, 2021-05-06:
@swagatk Were you able to fix the issue? When I switch to using the VelocityJointInterface the robot sags. I am able to send velocity commands to the robot but it continues to drop

https://user-images.githubusercontent.com/50928580/117355807-02726e00-ae81-11eb-93e3-34b3fce0a8a2.mp4

swagatk, 2021-05-06:
No, I could not resolve this issue. Please let me know if you are able to
solve this problem.

Thanks & Regards,
Swagat

sunnymints, 2021-05-17:
> @swagatk Were you able to fix the issue? When I switch to using the VelocityJointInterface the robot sags. I am able to send velocity commands to the robot but it continues to drop
> 
>  screencast_00001.mp4

I meet the problem too, I can not find this solution..... Were you able to fix the issue?

BAldhafeeri, 2022-06-02:
Remember `velocity_controllers/JointGroupVelocityController` is a forward controller so you don't need PID. You are sending the command directly (i.e. assuming your actuator accepts this kind of inputs). The error you get from Gazebo if you remove PID gains simply is not an error.