ros-industrial/universal_robot #478: Reuse moveit config files in custom setup?

jcvtieck, 2020-02-05:
Hello,
We have a bunch of UR robots in our lab (FZI Karlsruhe), and we are integrating moveit to our robot control pipeline.
In a normal use case, we have the robot mounted on a table and some kind of gripper.
What is the best way to use/generate/manage moveit config files here?
I mean, we have the same robot ex: UR5e in 3 different setups, the tables are different, and maybe different grippers or we test different ones, but the robot is the same.
The setup is described with a xacro that includes the official ur5e description.
For moveit, and motion planning, you will need to generate one moveit config with the MSA for every setup.
Is there a clever way of doing this?
I would like to reuse the moveit_ur5e_config files provided by you, and just change the transformations, and add the objects as obstacles.

any suggestions?

gavanderhoorn, 2020-02-05:
> I would like to reuse the moveit_ur5e_config files provided by you, and just change the transformations, and add the objects as obstacles.

I've written about this before, but I recommend you not do this.

Creating a MoveIt configuration package is almost trivial with the Setup Assistant.

As you already mentioned, they are essentially specific to specific robot and workcell configurations and only really work well if you generate them for these. Reusing existing ones doesn't really work well in my experience (and it's also not really worth it, with the MSA etc).

As to your comments:

> What is the best way to use/generate/manage moveit config files here?

I would still recommend to use the MSA, at least to get the basics set up.

> I mean, we have the same robot ex: UR5e in 3 different setups, the tables are different, and maybe different grippers or we test different ones, but the robot is the same.
> The setup is described [...]

I've seen people use `xacro` to generate / mutate the `srdf` and some of the other files. That is certainly possible, and would allow you to generate hw specific versions of the `srdf`. This could address your concern about having different packages "per gripper" (create a base one and insert the gripper info using `xacro`).

I can't point you to examples unfortunately.

---

Having written this, personally I really do just generate separate packages per hw configuration. I like to make things as clear and explicit as possible, and parameterising MoveIt configuration pkgs doesn't fit that approach. MoveIt configuration pkgs also haven't been designed with that in mind either, so it would also be an 'aftermarket' addition.

Customising MoveIt configuration pkgs also complicates updating them: customisations tend to break compatibility with the MSA.

gavanderhoorn, 2020-02-05:
As this is not really an issue with the packages in this repository, I'm closing this issue.

Feel free to keep commenting on it of course.

closed 2020-02-05 by gavanderhoorn