ros-industrial/universal_robot #482: loading urdf

ibrahim-kli, 2020-02-22:
[image]
I keep getting this issue I'm not sure what the error here is indicating, this occurs when I'm simply trying to load urdf files. 
I built catkin workspace because that was an issue , however, I'm stilling the same problem. 
any suggestion?
[image]

gavanderhoorn, 2020-02-23:
Please do not post screenshots of terminals. It's completely unnecessary (it's all text).

gavanderhoorn, 2020-02-23:
As to the rror you show: the top-level file to load would be `ur10_robot.urdf.xacro`.

`ur10.urdf.xacro` only defines a `xacro:macro`, but doesn't call/instantiate it, so after running it through `xacro`, you end up with an empty file.

Finally: you may want to use the provided [ur_description/launch/ur10_upload.launch](https://github.com/ros-industrial/universal_robot/blob/06d8b9e2f5f86aa54f9f2845f11edbc84e2f951e/ur_description/launch/ur10_upload.launch) to load the model to the parameter server (unless you're including it into some other macro, of course).

gavanderhoorn, 2020-02-23:
As this is not a problem with these packages, I'm going to close this issue.

Feel free to keep commenting of course.

closed 2020-02-23 by gavanderhoorn