ros-industrial/universal_robot #483: Installing of UR package

moa2020, 2020-02-27:
hi i am getting this problem any iea how to fix it 

CMake Error at universal_robot/ur_bringup/CMakeLists.txt:10 (find_package):
  By not providing "Findroslaunch.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "roslaunch", but CMake did not find one.

  Could not find a package configuration file provided by "roslaunch" with
  any of the following names:

    roslaunchConfig.cmake
    roslaunch-config.cmake

  Add the installation prefix of "roslaunch" to CMAKE_PREFIX_PATH or set
  "roslaunch_DIR" to a directory containing one of the above files.  If
  "roslaunch" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/moa/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/moa/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:318: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

gavanderhoorn, 2020-02-27:
I would suggest you post this sort of question on ROS Answers.

The issue tracker is for actual problem with the packages in this repository, not for user configuration problems / build environment setup.

As a short comment though: if you want to build things from source, you are responsible for installing all build dependencies.

But a system which is missing `roslaunch` while still having ROS installed is a bit of a strange situation.

gavanderhoorn, 2020-02-27:
As this is not an issue with the packages in this repository (or at least: right now I don't believe it is one), I'm going to close this issue.

Feel free to keep commenting on it.

closed 2020-02-27 by gavanderhoorn