ros-industrial/universal_robot #486: gazebo model and rviz model not aligned

rcywongaa, 2020-04-14:
In [`ur10.launch`](https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur_gazebo/launch/ur10.launch#L20), the gazebo model is spawned slightly offset, with the base `0.1` above the ground.  However, the `robot_state_publisher` still publishes a zero transform between `world` and `base`.  This causes the gazebo model and the rviz model to be misaligned.

gavanderhoorn, 2020-04-14:
That is certainly true -- and necessary, as the groundplane in Gazebo is not just a grid as in RViz, but an actual plane and will collide with the robot otherwise.

In practice, "the real world" is never aligned with your robot / ROS application. "zero" will always be where your application defines it. As Gazebo is a stand-in for the real world, all transforms will, as when using a real robot, be relative to the robot, not to a `world` frame necessarily. It's also the case that Gazebo's `world` frame does not coincide with the `world` frame in the xacro. 

---

Your issue seems to just report an observation, but fails to tell us about an actual problem. If you could clarify, we could see whether something needs to change, but right now it doesn't seem that way.

rcywongaa, 2020-04-15:
Thanks for the response.  My application involves a fixed 3D sensor looking at the robot arm.  Currently they share the same fixed frame `world` and rely on the TF tree to get their location w.r.t. the `world` frame.  The fact that the `ur10_robot.urdf.xacro` always publishes a zero `world` to `base_link` transform means TF tree always incorrectly represents the transform between the fixed `world` frame and the `base_link` frame.

I believe the following lines in `ur10_robot.urdf.xacro`
```
  <link name="world" />

  <joint name="world_joint" type="fixed">
    <parent link="world" />
    <child link = "base_link" />
    <origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0" />
  </joint>
```
should not be there as it should be up to the caller to determine the `world -> base_link` transform.  That way, if the model is spawned 0.1m above ground (or whatever position the robot arm is placed, w.r.t. to a fixed frame in the real world), this can be properly represented as a user-defined `world->base_link` transform.

Let me know what you think.

closed 2020-04-16 by rcywongaa