ros-industrial/universal_robot #488: Reason of zero damping and friction in URDF

gokhansolak, 2020-04-16:
In the xacro/urdf files, the damping and friction coefficients are both given as 0 for each joint. Is this because the UR drivers handle these implicitly, or they are too low that they are not important, or ignored for another reason?

This is not causing any problems, I am asking just to increase my understanding of the package.

Thank you in advance.

gavanderhoorn, 2020-04-16:
Those are not used anywhere, except in simulation.

Gazebo for instance requires those elements, otherwise it will ignore the joints.

With real hw those are not used.

gavanderhoorn, 2020-04-16:
And as this is not an issue with the packages in this repository, I'm going to close the issue.

Feel free to keep commenting of course.

closed 2020-04-16 by gavanderhoorn