ros-industrial/universal_robot #491: Any tutorial on Windows installation?

antonywu, 2020-05-18:
Attemping to install UR package for ROS on Windows, but I wasn't able to move past the following error.

`C:\ws>rosdep install --rosdistro melodic --ignore-src --from-paths src

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ur_kinematics: No definition of [moveit_ros_planning] for OS [windows]
ur10_moveit_config: No definition of [moveit_simple_controller_manager] for OS [windows]
ur5_moveit_config: No definition of [moveit_simple_controller_manager] for OS [windows]
ur3_e_moveit_config: No definition of [moveit_simple_controller_manager] for OS [windows]
ur3_moveit_config: No definition of [moveit_simple_controller_manager] for OS [windows]
ur10_e_moveit_config: No definition of [moveit_simple_controller_manager] for OS [windows]
ur5_e_moveit_config: No definition of [moveit_simple_controller_manager] for OS [windows]`

I understand that I would need to compile from source, but what are the steps to get the compilation going? Is there a tutorial available?

Thanks

gavanderhoorn, 2020-05-18:
No. I'm afraid not.

We don't really support Windows with this package.

For the MoveIt configurations you'll have to have MoveIt installed.

Without that it's not going to work.

gavanderhoorn, 2020-06-06:
As it seems this is blocked on an external package it doesn't seem actionable here.

As of today (2020-06-06) we don't officially support Windows with these packages, so non-working things are to be expected.

@antonywu: if you've figured this out in the meantime, I'd be interested to hear what you did.

gavanderhoorn, 2020-06-06:
We'll revisit Windows support in the near future.

I've had some more requests for this.

closed 2020-06-06 by gavanderhoorn