ros-industrial/universal_robot #493: How to real-time control ur5 end speed in gazebo7

richardsanqi, 2020-06-01:
I want to gazebo7 to control the speed of ur5 end by code,  Do you have a specific example or program .I am in this trouble for a long time.

gavanderhoorn, 2020-06-01:
This is not really an issue specific to the packages in this repository: I would advise you to look for a Gazebo plugin / `gazebo_ros_control` plugin which supports Cartesian velocity control.

If you find one, you'll only have to configure the model in `ur_gazebo` to use it.

That should be it.

I'm not aware of such a plugin (or: it has been too long since I've done anything with this).

gavanderhoorn, 2020-06-01:
As this is not really an issue with the package(s) in this repository, I'm going to close it.

Feel free to keep commenting on it of course.

closed 2020-06-01 by gavanderhoorn