ros-industrial/universal_robot #494: No planning library

mremson, 2020-06-05:
Hi,
I have installed ros-melodic on Ubuntu 18.04 and built the universal_robot package from source as described in the repository https://github.com/ros-industrial/universal_robot.
I already tried launching the joint limited version of ur5 but I still get the RuntimeError: Unable to connect to move_group action server 'move_group' within allotted time (30s). And I assume this causes that there is no planning library loaded.

I am running the following commands, each in a different command window:

roslaunch ur_gazebo ur5.launch limited:=true

roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true limited:=true

roslaunch ur5_moveit_config moveit_rviz.launch config:=true

jiannan0721, 2021-08-06:
Hello, did you resolve this issue? We encountered the same error.

lianghongzhuo, 2021-08-06:
use @fmauch 's branch https://github.com/fmauch/universal_robot/tree/calibration_devel
and `roslaunch ur_gazebo ur5_bringup.launch limited:=true`
This repo is far behind the state of the art.

jiannan0721, 2021-08-06:
> use @fmauch 's branch https://github.com/fmauch/universal_robot/tree/calibration_devel
> and `roslaunch ur_gazebo ur5_bringup.launch limited:=true`
> This repo is far behind the state of the art.

Thank you for the quick response! We will try.

ipa-nhg, 2021-09-29:
This will be solved once #538 is merged