ros-industrial/universal_robot #510: description: verify joint limits

gavanderhoorn, 2020-06-16:
Currently doing this for `melodic-devel-staging` and noticed that the joint limits for the UR5 fi are probably incorrect (at least the effort limits seem to be incorrect).

I'm slightly unsure what to use as effort limits though: [the docs](https://www.universal-robots.com/articles/ur-articles/max-joint-torques) state this:

> Note that it is the joint itself that has these torque values. You cannot use the full value in your application

So what should we configure as limits then?

gavanderhoorn, 2020-06-16:
@ipa-nhg @miguelprada @fmauch thoughts?

fmauch, 2020-06-16:
I think, it does make sense to use those as the full torque limits. As I understand it, in most cases a total tcp (or elbow) force will be limiting those maximum torques depending on the robot's joint configuration.

So, if we would want to stop users from driving the robot into safety limits by requesting high torques, we would also have to specify a force limit with a Cartesian reference and make sure this would be similar to the one configured on the robot.

Additionally, in the real world they will probably be hard to violate in a non-contact situation, as commanded jerks will be smoothed by the underlying `servoj` or `speedj` commands.

For in-contact situations, I think, it is not really unexpected that the robot goes into a protective stop, if the force limits are reached, although joint torque limits aren't met, yet.

gavanderhoorn, 2020-06-17:
I've used the listed torque limits as the maximum limits in the configuration files in the end.

Thanks for your comment @fmauch :+1:

gavanderhoorn, 2020-06-17:
Closing as #513 was merged.

closed 2020-06-17 by gavanderhoorn