ros-industrial/universal_robot #515: [ur_description] fix inconsistent namespace redefinitions for xmlns:xacro

fmessmer, 2020-06-17:
I've seen this for a while now:
```
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://www.ros.org/wiki/xacro
 new: http://wiki.ros.org/xacro (/home/....../src/universal_robot/ur_description/urdf/ur_macro.xacro)
```

gavanderhoorn, 2020-06-17:
Which version of ROS?

Namespaces are not carried over any more by `xacro` to the generated URDF, so you should not get this warning.

fmessmer, 2020-06-17:
I'm still on `kinetic`

gavanderhoorn, 2020-06-17:
It was fixed in `1.13.3`, which is the Melodic version ([status_page/ros_melodic_default.html?q=xacro](http://repositories.ros.org/status_page/ros_melodic_default.html?q=xacro)).

Kinetic won't get this fix ([status_page/ros_kinetic_default.html?q=xacro](http://repositories.ros.org/status_page/ros_kinetic_default.html?q=xacro)).

gavanderhoorn, 2020-06-17:
As it's only a warning and has no actual consequences (as in: the resulting urdf works), we won't be fixing this.

I'd be willing to merge a PR against `melodic-devel-staging` which essentially reverts #398, but do read the comments there before you do.

fmessmer, 2020-06-17:
ok, but if the `xacro=1.13.3` does not care about the namespace anymore - we could still get rid of this warning for kinetic by replacing the mentioned NEW namespace, right?

gavanderhoorn, 2020-06-17:
Closing as this is essentially a `works-for-me`.

If you feel that's not correct, we can re-open.

gavanderhoorn, 2020-06-17:
> ok, but if the `xacro=1.13.3` does not care about the namespace anymore - we could still get rid of this warning for kinetic by replacing the mentioned NEW namespace, right?

Please read the extensive discussion on #398. There is no "good" solution to this.

closed 2020-06-17 by gavanderhoorn