ros-industrial/universal_robot #522: Add a 'tool0_controller' frame to ur_gazebo [discussion]

gavanderhoorn, 2020-07-01:
As per subject.

Both `ur_modern_driver` and `ur_robot_driver` broadcast a `tool0_controller` frame, which is a direct copy of the values broadcast by the UR controller.

There is value in having this frame in the Gazebo simulation package as well, as it would make running the same software with both a real controller as well as Gazebo easier.

The simulation does not include a controller, so this would mean that `tool0 == tool0_controller`, but the frame would be there, and could be used by other ROS nodes.

gavanderhoorn, 2020-07-01:
This is a relatively trivial thing to add: simply add the `link` to `ur_gazebo/urdf/ur_macro.xacro` and attach it to `tool0` using a `fixed` `joint`.

fmauch, 2022-11-08:
Closed via #611

closed 2022-11-08 by fmauch