ros-industrial/universal_robot #523: Force/Impedance control

deepakraina99, 2020-07-06:
Is it possible to do cartesian impedance control using ROS for UR e-series? I am able to do with real robot using teach pendant but  I wanted it to do through ROS control? 
If not, do I have force feedback in UR series (in ROS topics) which can be used to write my own impedance controller?

Thanks.

gavanderhoorn, 2020-07-06:
There is no driver in this repository which you can use, so your question seems like it would be off-topic here.

Please post your question on a more appropriate issue tracker.

gavanderhoorn, 2020-07-06:
As this is not something which is actionable here for this repository, I'm going to close it.

If you feel this is incorrect, please clarify and we could re-open.

closed 2020-07-06 by gavanderhoorn