ros-industrial/universal_robot #527: ur3_moveit_config: setup assistant process dies

rfn123, 2020-07-13:
I am trying to edit the robot poses in the moveit config by using the setup assistant launch file (`roslaunch ur3_moveit_config setup_assistant.launch` ), however, when pressing the Load files button the progress bar goes to 60% and then the setup assistant crashes. 
gdb returns a segmentation fault:
```

[ INFO] [1594670856.700183317]: Loaded ur3 robot model.
[ INFO] [1594670856.701259296]: Setting Param Server with Robot Description
[ INFO] [1594670856.705576866]: Robot semantic model successfully loaded.
[ INFO] [1594670856.705609650]: Setting Param Server with Robot Semantic Description

Thread 1 "moveit_setup_as" received signal SIGSEGV, Segmentation fault.
__strlen_avx2 () at ../sysdeps/x86_64/multiarch/strlen-avx2.S:62
62	../sysdeps/x86_64/multiarch/strlen-avx2.S: No such file or directory.
(gdb) bt
#0  0x00007ffff5b0e811 in __strlen_avx2 ()
    at ../sysdeps/x86_64/multiarch/strlen-avx2.S:62
#1  0x00007ffff60afc4c in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::compare(char const*) const ()
    at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x00007ffff5756989 in moveit_setup_assistant::MoveItConfigData::inputPlanningContextLaunch(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
    at /home/usr/projects/catkin_ws/devel/lib/libmoveit_setup_assistant_tools.so.1.1.0
#3  0x00007ffff7b1b974 in moveit_setup_assistant::StartScreenWidget::loadExistingFiles() ()
    at /home/usr/projects/catkin_ws/devel/lib/libmoveit_setup_assistant_widgets.so.1.1.0
#4  0x00007ffff7b1c945 in moveit_setup_assistant::StartScreenWidget::loadFilesClick() ()
    at /home/usr/projects/catkin_ws/devel/lib/libmoveit_setup_assistant_widgets.so.1.1.0
#5  0x00007ffff65cb555 in QMetaObject::activate(QObject*, int, int, void**) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007ffff74bcba2 in QAbstractButton::clicked(bool) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#7  0x00007ffff74bcdba in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
---Type <return> to continue, or q <return> to quit---q
```

I am not sure if this is a moveit problem. I tried both binary install and building moveit from source, same error. 
Then I tried a custom moveit_config package and the setup assistant is not crashing.
I am using ROS melodic on Ubuntu 18.
Any hints how to solve this problem?

gavanderhoorn, 2020-07-14:
For now I'm going to assume this is a problem with the MSA or MoveIt in general.

I won't have time to debug this for at least the next two weeks, so I'd recommend you don't wait on that.

rfn123, 2020-07-14:
Ok, thanks for the reply!

gavanderhoorn, 2021-01-29:
I'm going to close this.

I don't know whether this is still an issue, but there hasn't been any activity on this since July last year.

We can always re-open.

closed 2021-01-29 by gavanderhoorn

VFjr, 2021-05-04:
I am getting the same exact error with any of the UR moveit configs. The setup assistant crashes at the same point. Was able to replicate it on two different PCs using ROS Melodic on Ubuntu 18.

I have checked the URDF files and there are no negative lengths in it.

dscho15, 2022-04-05:
I found the issue