ros-industrial/universal_robot #529: UR arms breaks - Warthog description update

tahir1069, 2020-07-14:
I am trying to update warthog description, by attaching three universal robot arms(2 UR5, 1 UR3). I am following the second  approach pointed [here](http://www.clearpathrobotics.com/assets/guides/kinetic/warthog/WarthogStartup.html#adding-urdf). I was succesfull in adding all the robot descriptions needed. But UR arm breaks which actually is included later. In the following example:
```
  <xacro:include filename="$(find mbs_description)/urdf/accessories/arms/right_ur_arm.urdf.xacro" />
  <xacro:if value="$(optenv WARTHOG_RIGHT_ARM 0)">
    <joint name="right_arm_joint" type="fixed">
      <origin xyz="0 0 -0.025" rpy="0 0 0" />
      <parent link="right_arm_mount_link"/>
      <child link="right_arm/base_link" />
    </joint>
  </xacro:if>

  <xacro:include filename="$(find mbs_description)/urdf/accessories/arms/left_ur_arm.urdf.xacro" />
  <xacro:if value="$(optenv WARTHOG_LEFT_ARM 0)">
    <joint name="left_arm_joint" type="fixed">
      <origin xyz="0 0 -0.025" rpy="0 0 0" />
      <parent link="left_arm_mount_link"/>
      <child link="left_arm/base_link" />
    </joint>
  </xacro:if>
  
  <xacro:include filename="$(find mbs_description)/urdf/accessories/arms/center_ur_arm.urdf.xacro" />
  <xacro:if value="$(optenv WARTHOG_CENTER_ARM 0)">
    <joint name="center_arm_joint" type="fixed">
      <origin xyz="0 0 -0.025" rpy="0 0 3.1415" />
      <parent link="center_arm_mount_link"/>
      <child link="center_arm/base_link" />
    </joint>
  </xacro:if>
```

Here I have added right and left (both UR5e) and than center(UR3), so the arm in the center breaks its joints and goes to the length of UR5e. 

And if I take the center arm at the top like here:

```
<xacro:include filename="$(find mbs_description)/urdf/accessories/arms/center_ur_arm.urdf.xacro" />
  <xacro:if value="$(optenv WARTHOG_CENTER_ARM 0)">
    <joint name="center_arm_joint" type="fixed">
      <origin xyz="0 0 -0.025" rpy="0 0 3.1415" />
      <parent link="center_arm_mount_link"/>
      <child link="center_arm/base_link" />
    </joint>
  </xacro:if>
  
  <xacro:include filename="$(find mbs_description)/urdf/accessories/arms/right_ur_arm.urdf.xacro" />
  <xacro:if value="$(optenv WARTHOG_RIGHT_ARM 0)">
    <joint name="right_arm_joint" type="fixed">
      <origin xyz="0 0 -0.025" rpy="0 0 0" />
      <parent link="right_arm_mount_link"/>
      <child link="right_arm/base_link" />
    </joint>
  </xacro:if>

  <xacro:include filename="$(find mbs_description)/urdf/accessories/arms/left_ur_arm.urdf.xacro" />
  <xacro:if value="$(optenv WARTHOG_LEFT_ARM 0)">
    <joint name="left_arm_joint" type="fixed">
      <origin xyz="0 0 -0.025" rpy="0 0 0" />
      <parent link="left_arm_mount_link"/>
      <child link="left_arm/base_link" />
    </joint>
  </xacro:if>
```

Now center arm will be fine but both  right and left will shrink to the length of center arm i.e. UR3. 

I am not able to sort out this issue, can anyone have an insight on the problem.

Note:

The robot descriptions [UR5e](https://github.com/fmauch/universal_robot/blob/calibration_devel/ur_e_description/urdf/ur5e_robot.urdf.xacro) and [UR3](https://github.com/fmauch/universal_robot/blob/calibration_devel/ur_description/urdf/ur3_robot.urdf.xacro) are the same as on the official [supported repo](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver)

Note: 
The same issue is raised on [Warthog's repo](https://github.com/warthog-cpr/warthog/issues/8) as currently I don't know where the problem could be.

tahir1069, 2020-07-14:
[image]
[image]

gavanderhoorn, 2020-07-16:
Please check whether the version of `ur_description` in `melodic-devel-staging` also exhibits this problem.

It's likely some parameters are not namespaced properly, causing last-setting-wins behaviour.

tahir1069, 2020-07-17:
Thank you so much for your comment @gavanderhoorn . 

I have changed the xacro file for each arm as in `melodic-devel-staging` according to `ur.xacro` but I get the following error.
```

[Errno 21] Is a directory: u'/home/student/warthog_data/TUD/warthog_ws/src/fmauch_universal_robot/ur_description/urdf/' 
when evaluating expression 'load_yaml(joint_limits_parameters_file)' 
when evaluating expression '__joint_limit_parameters['joint_limits']' 
when evaluating expression '__limits['shoulder_pan']['min_position']'
when instantiating macro: read_model_data (/home/student/warthog_data/TUD/warthog_ws/src/fmauch_universal_robot/ur_description/urdf/inc/ur_common.xacro)
instantiated from: ur_robot (/home/student/warthog_data/TUD/warthog_ws/src/fmauch_universal_robot/ur_description/urdf/ur_macro.xacro)
in file: /home/student/warthog_data/TUD/warthog_ws/src/fmauch_universal_robot/ur_description/urdf/ur.xacro
included from: /home/student/warthog_data/TUD/warthog_ws/src/mbs/mbs_description/urdf/accessories/arms/center_ur_arm.urdf.xacro
included from: /home/student/warthog_data/TUD/warthog_ws/src/mbs/mbs_description/urdf/accessories.urdf.xacro
included from: /home/student/warthog_data/TUD/warthog_ws/src/mbs/mbs_description/urdf/warthog.urdf.xacro
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/lib/warthog_description/env_run                   /home/student/warthog_data/TUD/warthog_ws/src/mbs/mbs_description/configs/mbs_warthog                   /opt/ros/kinetic/lib/xacro/xacro /home/student/warthog_data/TUD/warthog_ws/src/mbs/mbs_description/urdf/warthog.urdf.xacro                   --inorder] returned with code [2]. 

Param xml is <param command="$(find warthog_description)/scripts/env_run                   $(find mbs_description)/configs/$(arg config)                   $(find xacro)/xacro $(find mbs_description)/urdf/warthog.urdf.xacro                   --inorder" name="robot_description"/>
The traceback for the exception was written to the log file

```

Here is my current `xacro` file for one robot.

```
<?xml version="1.0"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro"
       name="ur3_robot" >

  <!-- ur -->
  <xacro:include filename="$(find ur_description)/urdf/ur.xacro" />

  <!--ur3 parameters files -->
  <xacro:arg name="joint_limit_params" default="$(find ur_description)/config/ur3/joint_limits.yaml"/>
  <xacro:arg name="kinematics_params" default="$(find ur_description)/config/ur3/default_kinematics.yaml"/>
  <xacro:arg name="physical_params" default="$(find ur_description)/config/ur3/physical_parameters.yaml"/>
  <xacro:arg name="visual_params" default="$(find ur_description)/config/ur3/visual_parameters.yaml"/>
  <!--common parameters -->
  <xacro:arg name="transmission_hw_interface" default="hardware_interface/PositionJointInterface" />
  <xacro:arg name="safety_limits" default="false" doc="If True, enable the safety limits controller"/>
  <xacro:arg name="safety_pos_margin" default="0.15" doc="The lower/upper limits in the safety controller" />
  <xacro:arg name="safety_k_position" default="20" doc="Used to set k position in the safety controller" />

  <xacro:ur_robot 
                prefix="center_arm/"
                joint_limits_parameters_file="$(arg joint_limit_params)"
                kinematics_parameters_file="$(arg kinematics_params)"
                physical_parameters_file="$(arg physical_params)"
                visual_parameters_file="$(arg visual_params)"
                transmission_hw_interface="$(arg transmission_hw_interface)"
                safety_limits="$(arg safety_limits)"
                safety_pos_margin="$(arg safety_pos_margin)"
                safety_k_position="$(arg safety_k_position)"
                />
</robot>

```

Sorry for asking an error here but I was not able to get where the problem could be?

gavanderhoorn, 2020-07-17:
I'm not sure what you've done exactly, as the error output from `xacro` mentions `fmauch_universal_robot`. If/when using `melodic-devel-staging`, that is not what I would expect.

This isn't going to be a simple copy-paste replacement I'm afraid. You can't take just a single file.

You must use the entire new `ur_description`.

gavanderhoorn, 2020-07-17:
Perhaps UniversalRobots/Universal_Robots_ROS_Driver#97 can help.

gavanderhoorn, 2020-07-17:
Oh and btw cross-posting to this repository, the CP one *and* ROS Answers seems like a bit too much.

I would recommend you pick one venue for diagnosing your issue and close the others (for now).

tahir1069, 2020-07-17:
> I'm not sure what you've done exactly, as the error output from `xacro` mentions `fmauch_universal_robot`. If/when using `melodic-devel-staging`, that is not what I would expect.
>
> This isn't going [...]

I have replaced the `ur_description` in `fmauch_universal_robot` from `melodic-devel-staging`, so I am using the new description. 

> Oh and btw cross-posting to this repository, the CP one _and_ ROS Answers seems like a bit too much.
> 
> I would recommend you pick one venue for diagnosing your issue and close the others (for now).

actually initially I was not sure where the problem would be so I cross posted by mentioning everything. But yes I will close the other ones by mentioning to this issue

gavanderhoorn, 2020-07-17:
> I have replaced the `ur_description` in `fmauch_universal_robot` from `melodic-devel-staging`, so I am using the new description.

This traceback seems to suggest something is still using the `fmauch_universal_robot` fork:

> I have changed the xacro file for each arm as in `melodic-devel-staging` according to `ur.xacro` but I get the following error.
> ```
> [Errno 21] Is a directory: [...]

ipa-nhg, 2020-07-22:
@tahir1069 which distro are you using? melodic or kinetic?

The [error](https://github.com/ros-industrial/universal_robot/issues/529#issuecomment-659939476) you copied says:

```
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/**kinetic**/lib/warthog_description/env_run                   /home/student/warthog_data/TUD/warthog_ws/src/mbs/mbs_description/configs/mbs_warthog                   /opt/ros/**kinetic**/lib/xacro/xacro /home/student/warthog_data/TUD/warthog_ws/src/mbs/mbs_description/urdf/warthog.urdf.xacro                   --inorder] returned with code [2]. 
```

I am not sure, but could be that some of the xacro expresion/tags from the `melodic-devel-staging` branch version were not supported for kinetic. Unfortunately I haven't got a kinetic installation to reproduce the error.

tahir1069, 2020-07-28:
I have figured out the issue. 
Actually I was importing `ur.xacro` where I was setting all the parameters. But I have to import `ur_macro.xacro` where after setting the correct parameters I was able to solve the above mentioned error.

tahir1069, 2020-07-28:
So problem here now is the same as mentioned above. This time the arms does not break as because of the same `ur_macro.xacro` but all of the models are same. If the _ur3_ is at the top, than all three arms will be _ur3_. As is the same case for _ur5e_ if it is at the top than all three arms will be the same i.e. _ur5e_

Here is the code responsible for all three arms:

```
 <!-- Arms -->
  
  <xacro:include filename="$(find mbs_description)/urdf/accessories/arms/right_ur_arm.urdf.xacro" />
  <xacro:if value="$(optenv WARTHOG_RIGHT_ARM 0)">
    <joint name="right_arm_joint" type="fixed">
      <origin xyz="0 0 -0.025" rpy="0 0 0" />
      <parent link="right_arm_mount_link"/>
      <child link="right_arm/base_link" />
    </joint>
  </xacro:if>

  <xacro:include filename="$(find mbs_description)/urdf/accessories/arms/left_ur_arm.urdf.xacro" />
  <xacro:if value="$(optenv WARTHOG_LEFT_ARM 0)">
    <joint name="left_arm_joint" type="fixed">
      <origin xyz="0 0 -0.025" rpy="0 0 0" />
      <parent link="left_arm_mount_link"/>
      <child link="left_arm/base_link" />
    </joint>
  </xacro:if>
  
  <xacro:include filename="$(find mbs_description)/urdf/accessories/arms/center_ur_arm.urdf.xacro" />
  <xacro:if value="$(optenv WARTHOG_CENTER_ARM 0)">
    <joint name="center_arm_joint" type="fixed">
      <origin xyz="0 0 -0.025" rpy="0 0 3.1415" />
      <parent link="center_arm_mount_link"/>
      <child link="center_arm/base_link" />
    </joint>
  </xacro:if>
```

and for all three center_arm, right_arm and left_arm the configuration is the same as in the [launch files](https://github.com/ros-industrial/universal_robot/tree/melodic-devel-staging/ur_description/launch), only included in xacro xml format.

```
<?xml version="1.0"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro"
       name="ur3_robot" >

  <!-- ur -->
  <xacro:include filename="$(find ur_description)/urdf/ur_macro.xacro" />

  <!--ur3 parameters files -->
  <xacro:arg name="joint_limit_params" default="$(find ur_description)/config/ur3/joint_limits.yaml"/>
  <xacro:arg name="kinematics_params" default="$(find ur_description)/config/ur3/default_kinematics.yaml"/>
  <xacro:arg name="physical_params" default="$(find ur_description)/config/ur3/physical_parameters.yaml"/>
  <xacro:arg name="visual_params" default="$(find ur_description)/config/ur3/visual_parameters.yaml"/>
  <!--common parameters -->
  <xacro:arg name="transmission_hw_interface" default="hardware_interface/PositionJointInterface" />
  <xacro:arg name="safety_limits" default="false" doc="If True, enable the safety limits controller"/>
  <xacro:arg name="safety_pos_margin" default="0.15" doc="The lower/upper limits in the safety controller" />
  <xacro:arg name="safety_k_position" default="20" doc="Used to set k position in the safety controller" />

  <xacro:ur_robot 
                prefix="center_arm/"
                joint_limits_parameters_file="$(arg joint_limit_params)"
                kinematics_parameters_file="$(arg kinematics_params)"
                physical_parameters_file="$(arg physical_params)"
                visual_parameters_file="$(arg visual_params)"
                transmission_hw_interface="$(arg transmission_hw_interface)"
                safety_limits="$(arg safety_limits)"
                safety_pos_margin="$(arg safety_pos_margin)"
                safety_k_position="$(arg safety_k_position)"
                />
</robot>

```

For both right and left _ur5e_ are there so there xacro descriptions are the same only with different prefixes:

```
<?xml version="1.0"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro"
       name="ur5e" >

  <!-- ur -->
  <xacro:include filename="$(find ur_description)/urdf/ur_macro.xacro" />

  <!--ur5e parameters files -->
  <xacro:arg name="joint_limit_params" default="$(find ur_description)/config/ur5e/joint_limits.yaml"/>
  <xacro:arg name="kinematics_params" default="$(find ur_description)/config/ur5e/default_kinematics.yaml"/>
  <xacro:arg name="physical_params" default="$(find ur_description)/config/ur5e/physical_parameters.yaml"/>
  <xacro:arg name="visual_params" default="$(find ur_description)/config/ur5e/visual_parameters.yaml"/>
  <!--common parameters -->
  <xacro:arg name="transmission_hw_interface" default="hardware_interface/PositionJointInterface" />
  <xacro:arg name="safety_limits" default="false" doc="If True, enable the safety limits controller"/>
  <xacro:arg name="safety_pos_margin" default="0.15" doc="The lower/upper limits in the safety controller" />
  <xacro:arg name="safety_k_position" default="20" doc="Used to set k position in the safety controller" />
  
  <xacro:ur_robot 
                prefix="right_arm/"
                joint_limits_parameters_file="$(arg joint_limit_params)"
                kinematics_parameters_file="$(arg kinematics_params)"
                physical_parameters_file="$(arg physical_params)"
                visual_parameters_file="$(arg visual_params)"
                transmission_hw_interface="$(arg transmission_hw_interface)"
                safety_limits="$(arg safety_limits)"
                safety_pos_margin="$(arg safety_pos_margin)"
                safety_k_position="$(arg safety_k_position)"                
                />

  />
</robot>

```

gavanderhoorn, 2020-07-28:
I'm going to have to ask you to create an [SSCCE](http://sscce.org), as having to go through all the Clearpath xacros and `.launch` files is not doable.

Once you have an SSCCE, post a link to it here and we could take a look.

tahir1069, 2020-07-28:
I am creating the SSCCE here, so that it can help others in future to benefit. 

I will share each and every step with files to recreate the issue. 

```
mkdir -p test_ws/src && cd test_ws/src 

git clone https://github.com/ros-industrial/universal_robot/

cd universal_robot

git checkout melodic-devel-staging

cd ..

catkin_create_pkg test_urdf

# Extract robot_model.tar.gz to test_urdf package just created. 

cd ..

catkin build 

source devel/setup.bash

roslaunch test_urdf model.launch
```
[robot_model.tar.gz](https://github.com/ros-industrial/universal_robot/files/4987740/robot_model.tar.gz)

Currently all three arms will be `ur5e` because `right_arm` is at the top in `test.urdf.xacro`. Just taking `center_arm` to the top will make all three arms to `ur3`

gavanderhoorn, 2020-07-28:
The files in `urdf/arms` are not macros, but top-level `.xacro`s. You cannot `xacro:include` those in another file.

Can you clarfy why you create those files? They appear not to be needed, as you could "just" `xacro:include` the `ur_macro.xacro` once, and then instantiate the model three times, supplying the appropriate arguments to the `xacro:ur_robot` macro.

gavanderhoorn, 2020-07-28:
I'm going to assume the `xacro` macros in this repository are actually OK, and the reported behaviour is the result of a configuration error on the user side for now.

I'll also close this issue in anticipation of confirmation of this by @tahir1069.

If it turns out something is actually wrong here in `universal_robot`, we can re-open.

closed 2020-07-28 by gavanderhoorn

gavanderhoorn, 2020-07-30:
@tahir1069: could you please confirm `xacro:include`ing the correct files fixes the issue for you?

Note: during WRID20 we had a discussion about creating "convenience" macros which would provide the same defaults as the ones set in the top-level `.xacro`, but per-variant. See https://github.com/ros-industrial/universal_robot/compare/melodic-devel-staging...gavanderhoorn:convenience_macros for a WIP.

tahir1069, 2020-08-03:
> The files in `urdf/arms` are not macros, but top-level `.xacro`s. You cannot `xacro:include` those in another file, as "the last one" will essentially overwrite the values for all the `xacro:arg`s of the others -- which seems to explain what you report.

Yes this was the issue. I actually didn't knew that this would happen because they were in seperate files. But as they are included in a top level file so essentially this should over write the values. But still the the solution suggested by you worked out and the model is fixed.

> Can you clarify why you create those files?

To have everything separate in the repository and include the ones dependent upon the environment variables.
 

> @tahir1069: could you please confirm `xacro:include`ing the correct files fixes the issue for you?

Yes indeed!

> See [melodic-devel-staging...gavanderhoorn:convenience_macros](https://github.com/ros-industrial/universal_robot/compare/melodic-devel-staging...gavanderhoorn:convenience_macros) for a WIP.

Exactly this is what I tried to do and now its working. 

Thank you so much for your help so far.

tahir1069, 2020-08-04:
I am having a small issue similar to this one in Gazebo. @gavanderhoorn  may be you can point out to the problem!

```
mkdir -p test_ws/src && cd test_ws/src 

git clone https://github.com/ros-industrial/universal_robot/

catkin_create_pkg test_urdf

# Extract test_urdf.zip(attached to this comment) to test_urdf package just created. 

cd ..

catkin_make

source devel/setup.bash

roslaunch test_urdf ur_gazebo.launch
```
[test_urdf.zip](https://github.com/ros-industrial/universal_robot/files/5022189/test_urdf.zip)

I can only move one arm when execute the following command

> rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller

I have properly namespaced the nodes and tf_prefixes. I can also verify in _rqt_graph_ the respective node connections when changing the controller in _rqt_joint_trajectory_controller_.

All other controllers seems to be launched and I can change the value in _rqt_joint_trajectory_controller_ but nothing moves apart from _left_arm_