ros-industrial/universal_robot #530: Rviz not displaying UR5 model properly - Melodic - Ubuntu 18.04

BEC17, 2020-07-24:
Hello,

I've been using UR5 packages to follow a tutorial but I found some issues. First of all, when I execute my launch file, I notice that the UR5 model is not completed, as it can be seen in the picture:
[image]
I also have attached my xacro file but I think that there is no problem with that, nevertheless...
`<?xml version="1.0"?>
<robot name="my_robot" xmlns:xacro="http://ros.org/wiki/xacro">

    <xacro:include filename="$(find ur_description)/urdf/ur5.urdf.xacro" />
    <xacro:ur5_robot prefix="robot1_" joint_limited="true"/>

    <link name="world"/>
    

    <link name="table">
        <visual>
          <geometry>
            <box size="1 1 0.05"/>
          </geometry>
        </visual>
        <collision>
          <geometry>
            <box size="1 1 0.05"/>
          </geometry>
        </collision>
    </link>

    <joint name="world_to_table" type="fixed">
      <parent link="world"/>
      <child link="table"/>
      <origin xyz="0 0 1" rpy="0 0 0"/>
    </joint>

    

    <joint name="table_to_robot" type="fixed">
      <parent link="table"/>
      <child link="robot1_base_link"/>
      <origin xyz="0 0 0" rpy="0 0 0"/>
    </joint>

</robot>`
I have already checked out the solution proposed in #374 but never happened, everything remains the same.
I have also noticed in Rviz that in Fixed Frame, there is no selection option in the drop down.

I have also tried to use the following commands in order to identify something that help me out with my main issue
`roslaunch ur_gazebo ur5.launch   
roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true
roslaunch ur5_moveit_config moveit_rviz.launch config:=true`

I have identified the following error in the log message:
`roslaunch ur_gazebo ur5.launch   
roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true
roslaunch ur5_moveit_config moveit_rviz.launch config:=true`

and this warning:
`[ WARN] [1595553784.205487531]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ WARN] [1595553784.276379674]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/robot_description_kinematics/manipulator/kinematics_solver_attempts' from your configuration.
`

I do not know how it can be related to my main issue... moreover when I run `roslaunch ur5_moveit_config moveit_rviz.launch config:=true` UR5 model try to reach a random goal but it is not possible for it as seen in #427. I have also found that the solution for that issue is the same presented in #374 but it did not work for me, I do not know why.

If you have any suggestion, I really appreciate it. Thank you in advance.

gavanderhoorn, 2020-07-24:
> I have identified the following error in the log message:
> `roslaunch ur_gazebo ur5.launch roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true roslaunch ur5_moveit_config moveit_rviz.launch config:=true`

did you intent to copy an error message here?

BEC17, 2020-07-25:
yes, sorry. Here it is:

`[ERROR] [1595635891.890535236, 0.001000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/shoulder_pan_joint

[ERROR] [1595635891.892863547, 0.001000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/shoulder_lift_joint

[ERROR] [1595635891.896129059, 0.001000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/elbow_joint

[ERROR] [1595635891.898482817, 0.001000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_1_joint

[ERROR] [1595635891.902553719, 0.001000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_2_joint

[ERROR] [1595635891.904936590, 0.001000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_3_joint
`

gavanderhoorn, 2020-07-25:
That's unrelated.

Please also format code blocks correctly ([docs](https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks)).

BEC17, 2020-07-25:
> That's unrelated.
> 
> Please also format code blocks correctly ([docs](https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks)).

What you mean with unrelated? In your previous reply you asked for an error in the log message then I copied the error, clarifying what I meant in my first comment

gavanderhoorn, 2020-07-28:
@BEC17 wrote:
> What you mean with unrelated?

the `ERROR` about the PID gains not present. That's unrelated to the issue you describe -- or at least: very unlikely to be related.

gavanderhoorn, 2020-07-28:
Seeing your RViz screenshot: do you have a (static) transform broadcaster for the `map` frame? I ask as `map` is not a frame which is part of the URDFs, nor is it published by anything related.

If not: please use the dropdown to make another frame the root -- one which does have a publisher.

BEC17, 2020-07-29:
> Seeing your RViz screenshot: do you have a (static) transform broadcaster for the `map` frame? I ask as `map` is not a frame which is part of the URDFs, nor is it published by anything related.
> 
> If not: please use the dropdown to make another frame the root -- one which does have a publisher.

There is no selection option in the dropdown, to use a different frame I have to write it down by my own (e.g. robot1_base_link)

[image]

as you can see in the image, there is no change in the robot model

gavanderhoorn, 2020-07-29:
Then it would appear you have no TFs being broadcast at all.

What confuses me in your screenshot is that you show RViz, but it's not the RViz I would expect when you `roslaunch ur5_moveit_config moveit_rviz.launch config:=true`, as that would show the MoveIt RViz plugin, and additionally, *ROS Time* shown in your screenshot also does not appear to be simulation time (which I would expect is used as you say you `roslaunch ur_gazebo ur5.launch`.

I'm pretty sure there isn't anything wrong with the packages in this repository (or at least not in a way which could cause this). I'm inclined to believe this is user-error at this point.

Could you please tell us (again) which sequence of commands leads to the RViz as you show it in your latest screenshot.

Please show verbatim copies of the commands, associated / resulting console output and please attach a screenshot of `rqt_graph` and `rqt_tf_tree` after you've started everything.

ipa-nhg, 2020-07-29:
I am agree with @gavanderhoorn it looks like an error with tf (and/or the robot_state_publisher).

you can try the following two commands to debug the problem:

`rostopic echo /tf `
-> Is the topic publishing the tree?

`rosnode info /robot_state_publisher`
-> is the node running, subscribing /joint_states and publishing /tf ?

BEC17, 2020-07-30:
> Could you please tell us (again) which sequence of commands leads to the RViz as you show it in your latest screenshot.

`source $HOME/catkin_ws/devel/setup.bash`
`roslaunch my_robot_description myrobot.launch`

BEC17, 2020-07-30:
> associated / resulting console output and please attach a screenshot of `rqt_graph` and `rqt_tf_tree` after you've started everything.

This is the output of `roslaunch my_robot_description myrobot.launch`:

```
SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.6

NODES
  /
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [6984]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d58e384c-d208-11ea-b01b-74dfbfacd421
process[rosout-1]: started with pid [6995]
started core service [/rosout]
process[rviz-2]: started with pid [7002]
^C[rviz-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```
Screenshot of `rqt_graph`:

[image]

I must mention that I unchecked the debug option, otherwise I would not have gotten any output

Screenshot of `rqt_tf_tree`:

[image]

BEC17, 2020-07-30:
> And: seeing as you're trying to load a custom URDF/xacro: _which_ file(s) have you edited to load your file, instead of the default?

I only created two files in my catkin space; one is showed in my first comment and the other one was a launch file (the one I copy below)

```<launch>
  <arg name="gui" default="true"/>
  <param name="robot_description" command="$(find xacro)/xacro --inorder '$(find my_robot_description)/urdf/my_robot.xacro'"/>
  <node name="rviz" pkg="rviz" type="rviz" if="$(arg gui)"/>
</launch>
```

BEC17, 2020-07-30:
> I am agree with @gavanderhoorn it looks like an error with tf (and/or the robot_state_publisher).
>
> you can try the following two commands to debug the problem:
>
> `rostopic echo /tf `
> -> Is the topic [...]

There is no output for `rostopic echo /tf `...

This is the output for `rosnode info /robot_state_publisher`

```
Node [/robot_state_publisher]
Publications: None

Subscriptions: None

Services: None

cannot contact [/robot_state_publisher]: unknown node
```

gavanderhoorn, 2020-07-30:
`rqt_tf_tree` seems to confirm no TF is broadcast at all. That would be the immediate cause of "Rviz not displaying UR5 model properly".

However, it would have been more correct to write: "Rviz does not display my custom URDF which instantiates a UR5 from this repository", as that is what it seems you are actually trying to do.

Note: this is the first time you mention you have created a separate `.launch` file, and the first time you tell us that's where the RViz instance you show in your screenshots comes from. This is important information, and you should have included that in your OP.

It's unclear to me whether you try to run your `.launch` file *after* having started the files we provide here, or *instead* of those files.

If it's the former: that would overwrite the `robot_description` parameter and then start another RViz instance. I'm not sure whether that will work (ie: RViz would see a `robot_description` with your URDF in it, while all the previously started nodes would use the plain UR5 model). I would expect TF problems at least then.

If the latter: if what you show is really everything you have in your `.launch` file, that would be the cause of your problems. You'll have to add a `robot_state_publisher` and probably a `joint_state_publisher` at the very least. Without the first, no TF will be broadcast, but it will depend on the second to receive (fake) `JointState`s as input to its FK.

Finally:

> I've been using UR5 packages **to follow a tutorial** but I found some issues. 

Seeing as we have no tutorials which cover the topics you appear to be working on, this would be an external resource.

I would suggest you contact the author(s) of that tutorial and ask them what is going wrong.

As I believe the files we provide here are in a working state (but I'm certainly not claiming they are necessarily fault free), I'll restate my assumption this is a configuration/usage problem on your side, and not something we need to fix here.

As a result, I'm going to close your issue. Not because I'm dismissing the fact you are encountering difficulties getting your URDF to visualise properly, but because this doesn't appear to be caused by anything we (don't) do here, and thus this issue tracker is not the correct venue to ask for support.

gavanderhoorn, 2020-07-30:
As always, please feel free to keep commenting on the issue.

If later it turns out something is amiss here, we can re-open.

closed 2020-07-30 by gavanderhoorn

BEC17, 2020-07-31:
> If the latter: if what you show is really everything you have in your `.launch` file, that would be the cause of your problems. You'll have to add a `robot_state_publisher` and probably a [...]

Thank you for your help. I have added `robot_state_publisher` and `joint_state_publisher` to my `.launch` file and it worked as you can see in the image.

[image]

I know the issue is already closed but may you please tell me how I can reset the `robot_description` parameter? and finally, could you recommend me a good tutorial about ROS-Industrial and more specifically the Universal_Robot package.

Thanks again!

gavanderhoorn, 2020-07-31:
> how I can reset the `robot_description` parameter

I'm not sure I understand.

What exactly do you want to do?

> could you recommend me a good tutorial about [..] the Universal_Robot package.

I don't believe those exist.

All robot support packages in ROS-Industrial follow the same structure, and can be used in almost the same way.

This allows you to reuse your experience across different OEMs.

Afaik there are no tutorials specific to any one OEM.

> could you recommend me a good tutorial about ROS-Industrial

You could take a look at the [ROS-Industrial Training Exercises](https://industrial-training-master.readthedocs.io/en/melodic/index.html).

BEC17, 2020-08-01:
> I'm not sure I understand.
> 
> What exactly do you want to do?
 
You wrote in your previous comment "RViz would see a `robot_description` with your URDF in it, while all the previously started nodes would use the plain UR5 model"; wouldn't it be necessary to reset the `robot_description` parameter? I know it did not happen but the doubt came to mind when I was looking for information related to your answer; besides I found a tutorial in which was used the following file `reset_robot_description.launch`, then I was looking into the Universal_Robot package and did not found a file the one mentioned before.

BEC17, 2020-08-01:
> You could take a look at the [ROS-Industrial Training Exercises](https://industrial-training-master.readthedocs.io/en/melodic/index.html).

Thank you for the info. about the tutorial

gavanderhoorn, 2020-08-01:
> You wrote in your previous comment "RViz would see a `robot_description` with your URDF in it, while all the previously started nodes would use the plain UR5 model"; wouldn't it be necessary to [...]

You still don't tell us what you want to achieve, only how you believe you should try to achieve it.

*Why* do you believe `robot_description` needs to be reset?

Regular parameters are seldomly changed after they've been initialised. Especially `robot_description`.

> besides I found a tutorial in which was used the following file `reset_robot_description.launch`, 

"I found a tutorial" -> always link to something you refer to. We cannot guess, nor can we know what you have in front of you.

> then I was looking into the Universal_Robot package and did not found a file the one mentioned before.

No, because such a file would not be needed afaict with normal use of the packages and infrastructure.

gavanderhoorn, 2020-08-01:
Could I ask you to please post questions about the *use* of the packages on ROS Answers?

We'd like to keep the issue tracker for reporting issues and discussing enhancements.

BEC17, 2020-08-02:
> Could I ask you to please post questions about the _use_ of the packages on ROS Answers?

Of course, I will do that. 

I am trying to learn ROS-Industrial and I thought it would be a good start for me anyway, I will follow your advice of ROS Answer. Finally, your responses have been very useful in these days.