ros-industrial/universal_robot #532: Question: Multi-robot Gazebo simulation configuration

tahir1069, 2020-08-11:
I am trying to simulate three Universal Robot arms in Gazebo. I was able to spawn all the robots in Gazebo but only the last robot(linked in the URDF) actually can move.  

To create the issue following steps can be followed:

    mkdir -p test_ws/src && cd test_ws/src 
    
    git clone https://github.com/ros-industrial/universal_robot/
    
    catkin_create_pkg test_urdf
    
    # Extract test_urdf.zip(attached to this post) to test_urdf package just created. 
    
    cd ..
    
    catkin_make
    
    source devel/setup.bash
    
    roslaunch test_urdf ur_gazebo.launch

    # In another terminal 

    rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller

[test_urdf.zip](https://github.com/ros-industrial/universal_robot/files/5027032/test_urdf.zip)

Currently after executing the above mentioned launch file a gazebo world with three arms will be simulated. Now only left arm can be moved as it is linked at the the end in `test_urdf/urdf/test.urdf.xacro`. If another arm is taken to the last in the same URDF than that can be moved only. 

The issue is similar to #529 which was also an issue of configuration. To me everything seems to be namespaced and I can verify the connection in `rqt_graph` when selecting the respective controller from `rqt_joint_trajectory_controller` gui.

ipa-nhg, 2020-08-12:
which branch are you using for `universal_robot`? I suggest to use the `melodic-devel-staging`

tahir1069, 2020-08-12:
I have developed everything for ROS Kinetic. Due to the robot official supoprt. Isn't it possible for kinetic-devel. Because it might break the system. Also on real robot ROS kinetic is installed.

ipa-nhg, 2021-09-29:
@tahir1069 is this issue still relevant? ROS kinetic is already EOL, and unfortunately we are not  able to give support anymore for that distro. 

Please re-open the issue if it is still there for melodic.

closed 2021-09-29 by ipa-nhg