ros-industrial/universal_robot #534: URDF - general question about the rotation about the coordinate frames

kathrindoerfler, 2020-08-31:
Regarding all robot descriptions in this repository: I am trying to understand why the conventions for the location and orientation of all axes of the links and joints in the robot descriptions for the UR robots (both for the e-series and standard series, see Image 1) do not follow the model given by [https://www.universal-robots.com/articles/ur/parameters-for-calculations-of-kinematics-and-dynamics/](https://www.universal-robots.com/articles/ur/parameters-for-calculations-of-kinematics-and-dynamics/)  (see Image 2)?

[image]
Image 1: URDF model and coordinate frames visualized in MoveIt of this repo

[image]
Image 2: robot model and coordinate frames as given by Universal Robots

@gavanderhoorn and all, has this been set once and never been changed, or is there a specific reason to why this does not coincide with the model given by UR? It would be great to know the background, and also, whether this could be changed in the future?

Many thanks for your answers!

fmauch, 2020-09-02:
This is currently changing. The `melodic-devel-staging` branch already contains a version where the frames are more similar to the original approach. However, that seems to cause [problems with other libraries such as kdl](https://answers.ros.org/question/359496/different-axes-of-rotation-arrangement-for-ur5/), so this will probably not be the final stage.

gavanderhoorn, 2020-09-02:
@kathrindoerfler: could you provide some more insight into what prompted you to post your question?

It's likely you have a specific reason, and it might turn out orientation of those intermediate frames doesn't actually matter (note: the `base` and `tool0` frames correspond to the *Base* feature on the controller and an all-zeros TCP (especially in `melodic-devel-staging`)).

gavanderhoorn, 2020-11-04:
I'm going to close this due to inactivity.

closed 2020-11-04 by gavanderhoorn