ros-industrial/universal_robot #535: Gazebo crash when plugin is added to ur10 urdf

alt0216, 2020-09-14:
I tried to add a gazebo model plugin inside the ur10_robot.urdf.xacro using <gazebo> and <plugin>.  However, every time I try to run it, it will generate the following error:
_[gazebo-2] process has died [pid 11147, exit code 139, cmd /home/jason4/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -u -e ode worlds/empty.world __name:=gazebo __log:=/home/jason4/.ros/log/c2a3a100-f6d8-11ea-b422-080027b07338/gazebo-2.log].
log file: /home/jason4/.ros/log/c2a3a100-f6d8-11ea-b422-080027b07338/gazebo-2*.log
[arm_controller_spawner-8] process has died [pid 11165, exit code 1, cmd /opt/ros/melodic/lib/controller_manager/controller_manager spawn arm_controller __name:=arm_controller_spawner __log:=/home/jason4/.ros/log/c2a3a100-f6d8-11ea-b422-080027b07338/arm_controller_spawner-8.log].
log file: /home/jason4/.ros/log/c2a3a100-f6d8-11ea-b422-080027b07338/arm_controller_spawner-8*.log
[joint_state_controller_spawner-7] process has died [pid 11160, exit code 1, cmd /opt/ros/melodic/lib/controller_manager/controller_manager spawn joint_state_controller __name:=joint_state_controller_spawner __log:=/home/jason4/.ros/log/c2a3a100-f6d8-11ea-b422-080027b07338/joint_state_controller_spawner-7.log].
log file: /home/jason4/.ros/log/c2a3a100-f6d8-11ea-b422-080027b07338/joint_state_controller_spawner-7*.log
[ros_control_controller_manager-9] process has died [pid 11167, exit code 1, cmd /opt/ros/melodic/lib/controller_manager/controller_manager load joint_group_position_controller __name:=ros_control_controller_manager __log:=/home/jason4/.ros/log/c2a3a100-f6d8-11ea-b422-080027b07338/ros_control_controller_manager-9.log].
log file: /home/jason4/.ros/log/c2a3a100-f6d8-11ea-b422-080027b07338/ros_control_controller_manager-9*.log_

What is the proper way to attach a gazebo model plugin to the robot arm?

gavanderhoorn, 2020-09-15:
To be honest I have no idea right now.

If I were to debug this, I'd create a very simple model and then try to incrementally add additional elements.

alt0216, 2020-09-16:
Thanks for replying. Do you know where I can attach the gazebo plugin? Do I do it at ur10.urdf.xacro or is there a specific place I should attach the gazebo plugin to.

ipa-nhg, 2020-10-08:
@alt0216 which version (i.e. branch) are you using?

Julien-Livet, 2021-09-29:
I have just forked the current package ([https://github.com/Julien-Livet/universal_robot](https://github.com/Julien-Livet/universal_robot)) and tested it on my machine (Noetic version). For testing, I just add in "ur10_robot.urdf.xacro" (in a proper way, include this file and call macro with following code in your own xacro) the following block (from [http://gazebosim.org/tutorials?tut=ros_gzplugins#Camera](http://gazebosim.org/tutorials?tut=ros_gzplugins#Camera)).
```
<!-- camera -->
<gazebo reference="ee_link">
...
```
but I don't observe any problem (Gazebo is running and /rrbot/camera1/image topic is working).
Have you always this problem (if no, can you close this issue) and if yes, can you describe more your setup ?

gavanderhoorn, 2021-09-29:
I'm going to assume this was either a transient problem or has since been fixed.

Thanks @Julien-Livet for testing it out on your system.

Closing as I believe this is no longer an issue.

closed 2021-09-29 by gavanderhoorn