ros-industrial/universal_robot #537: ROS Noetic and Ubuntu 20.04 support

Victorlouisdg, 2020-09-22:
Hi, I was wondering whether support for ROS Noetic and Ubuntu 20.04 is planned or if anyone is working on it. Also, what steps can I take to help?

gavanderhoorn, 2020-09-22:
A first step could be to see what the status is of [melodic-devel-staging](https://github.com/ros-industrial/universal_robot/tree/melodic-devel-staging) is under Noetic.

The MoveIt configs don't work, but I'm more interested in "the rest" at the moment.

Victorlouisdg, 2020-09-22:
Building the melodic-devel-staging branch works perfectly by following the instruction under _**Building from Source**_ in the readme. The only addition step I had to do is install rosdep with:

`sudo apt-get install python3-rosdep`

One problem I encountered is that the robots aren't properly visualized in RViz.

`roslaunch ur_description view_ur5.launch` results in this:

[image]

DLu, 2020-10-13:
This need a new release of xacro before the `!degrees` tags will work.

gavanderhoorn, 2020-10-13:
Has already been released.

DLu, 2020-10-13:
The noetic branch doesn't have this yet: https://github.com/ros/xacro/pull/252

gavanderhoorn, 2020-10-13:
That would be the *lagging* I mentioned.

DLu, 2020-10-13:
Yup, I figured. Just adding the reference.

gavanderhoorn, 2020-10-13:
https://github.com/ros/xacro/pull/252#issuecomment-707913128

gavanderhoorn, 2020-10-14:
https://github.com/ros/rosdistro/pull/26945

dqii, 2020-11-11:
@Victorlouisdg were you able to get Noetic / Ubuntu 20.04 to work? Or did you end up using Kinetic or Melodic?

Victorlouisdg, 2020-11-12:
@dqii I haven't had time to continue working with the UR10 yet. I will be working with it in the near future, but initially I'm going to stick with ROS Melodic. If that goes well, I might try getting this to work on Noetic again.

gavanderhoorn, 2020-11-12:
There hasn't been a Noetic sync since ros/rosdistro#26945 was merged (see [Discourse discussion](https://discourse.ros.org/t/can-we-get-a-noetic-sync-please/17240)), so the changes needed for `melodic-devel-staging` to work in Noetic have not yet reached the main repositories.

The testing repository would be needed to install the latest version of `xacro`, or otherwise a source-checkout of `ros/xacro` in your Catkin workspace would also work.

That should prevent the problem reported by @Victorlouisdg in https://github.com/ros-industrial/universal_robot/issues/537#issuecomment-696840518.

Victorlouisdg, 2020-12-02:
The Noetic sync is out ([link](https://discourse.ros.org/t/new-packages-for-noetic-2020-11-30/17641)), to get the newly released packages you can run: 

- `sudo apt update`
- `sudo apt upgrade`

The ur_description package seems to be working perfectly on Ubuntu 20.04 now :)

dqii, 2020-12-02:
Sorry just to clarify, which distribution did you use for this package? You were able to use it directly?

Victorlouisdg, 2020-12-05:
I'm using ROS Noetic on Ubuntu 20.04 and I tested the `melodic-devel-staging` branch. Note that I'm talking specifically about the `ur_description` package, not the other packages for the UR robots that are also contained in this repository.

gonzalocasas, 2021-03-19:
Is anyone working on getting the move_it_config packages fixed?

danzimmerman, 2021-05-07:
I'm having an issue on [RoboStack Noetic](https://github.com/RoboStack/ros-noetic) on Windows 10 with invisible Collada visual meshes on some of the robots (here UR5) from `melodic-devel-staging`:

[image]

See https://github.com/RoboStack/ros-noetic/issues/121 for details.

This doesn't seem to show up in more mainstream installations. My coworker's `ros-noetic-desktop-full` install on Ubuntu Focal has no problem with the visuals on the `melodic-devel-staging` UR5 robot (we did not yet try others). And some of the files (UR10) work perfectly on my RoboStack Noetic on Windows 10 as well. Zero issues on Windows or Linux with Melodic installations.

Even though it's a niche environment at the moment, I wanted to flag this just in case it might break in future from-source or binary builds on more mainstream ROS installations. I don't know if figuring out how to fix this via the structure of the `.dae` files is a correct approach or even a desirable workaround, but it seems to have something to do with Gazebo's material parsing, and doesn't affect the robots uniformly (I have noticed problems with the UR5e as well, haven't checked all of them yet)

Per my RoboStack issue above, I'm getting errors like

```
[Err] [..\gazebo\rendering\Visual.cc:374] Ogre Error:ItemIdentityException: Resource with the name robot::forearm_link::forearm_link_visual_MATERIAL_gazebo_material_2 already exists. in ResourceManager::add at ..\OgreMain\src\OgreResourceManager.cpp (line 158)
[Err] [..\gazebo\rendering\Visual.cc:375] Unable to create a mesh from C:/Code/ros/workspaces/noethp_ws/src\universal_robot\ur_description/meshes/ur5/visual/forearm.dae
[Err] [..\gazebo\rendering\Visual.cc:374] Ogre Error:ItemIdentityException: Resource with the name robot::upper_arm_link::upper_arm_link_visual_MATERIAL_gazebo_material_16 already exists. in ResourceManager::add at ..\OgreMain\src\OgreResourceManager.cpp (line 158)
[Err] [..\gazebo\rendering\Visual.cc:375] Unable to create a mesh from C:/Code/ros/workspaces/noethp_ws/src\universal_robot\ur_description/meshes/ur5/visual/upperarm.dae
```

Ideas for fixes or workarounds are appreciated.

danzimmerman, 2021-05-19:
Note that the invisible-links issue I described above is indeed an issue with how recent versions of OGRE handle materials (see  https://github.com/osrf/gazebo/issues/3002), not an issue with the models in these packages. It's now fixed in the Robostack version of Noetic and my `ur_gazebo` functionality is working there. I still have the same problem in RViz that @Victorlouisdg reports above... the robot is white and collapsed.

gavanderhoorn, 2021-05-20:
> I still have the same problem in RViz that @Victorlouisdg reports above... the robot is white and collapsed.

you don't show an RViz screenshot here, but from the one you show in https://github.com/conda-forge/gazebo-feedstock/issues/74#issuecomment-835426132:

[image]

It would appear you have the *Fixed Frame* setting set to `map`, and the TF display complains about no transforms being available from `map` to the frames of the robot model.

You'll need to change the *Fixed Frame* to some frame which does exist in your scene.

danzimmerman, 2021-05-20:
@gavanderhoorn Thanks. Makes sense re: the collapse with the TF errors. Is the featureless white visual appearance normal RViz behavior in this situation?

I am not very familiar with RViz because I've been working almost exclusively in Gazebo. Switching to a valid frame, I see that the behavior is identical on my ordinary ([MS-IoT](https://ms-iot.github.io/ROSOnWindows/)) Windows 10 Melodic installation and my Robostack Noetic installation:

Valid frame `world`
[image]

Missing frame `map`

[image]

sheila-suppiah, 2021-09-29:
@cardboardcode can you assign this to @hieutrungbk ?

gavanderhoorn, 2021-09-29:
This was assigned to @Hieutrungbk. Could you provide an update on what you found?

XiaoqZhan, 2022-01-10:
> Please clarify what you're going to be doing exactly.
> 
> Will you be using [UniversalRobots/Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver)? If so: please follow the instructions there. Do not use this repository by itself in that case.
> 
> Also: usage questions might be better posted on ROS Answers.

Excuse me, could you please explain the relationship between the [UniversalRobots/Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) and this repository?

gonzalocasas, 2022-01-10:
> Excuse me, could you please explain the relationship between the [UniversalRobots/Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) and this repository?

This repository contains the models (URDF/SRDF/etc) and the configuration for MoveIt, and the other one is one of the drivers for the real hardware.

aatb-ch, 2022-03-03:
Is there an update on the Noetic version?

I have tried the melodic-devel-staging branch, also fmauch moveit PR branch https://github.com/ros-industrial/universal_robot/pull/538 and nothing seems to be working for MoveIt launch files on Noetic. Wondering if anybody has UR10 moveit launch files working on Noetic.

Could you clarify also the conflicting infos when following the official steps to build [UniversalRobots/Universal_Robots_ROS_Driver ](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) which involves cloning fmauch's universal_robots calibration_devel branch, which then creates catkin build issues with duplicated packages in this repo? I've not found any info regarding that, if using this repo am I not supposed to clone fmauch's branch?

fmauch, 2022-03-03:
> 

Using my fork and the `calibration_devel` branch should not create duplicated packages unless you keep this repo in your workspace. Note that sometimes you'll have to clean your build folder if packages got replaced.

As of this day, the instructions on [UniversalRobots/Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) are still valid (and will be adapted in the future if things change).

fmauch, 2022-03-03:
@aatb-ch You should not clone both. No wonder you get the `Multiple packages found` error, as you have effectively cloned the same repo (in different versions) twice. **Only** use my fork (in your case delete  `~/catkin_ws/src/universal_robot`).

fmauch, 2022-03-03:
my calibration_devel branch **does** contain a workin moveit configuration.

aatb-ch, 2022-03-03:
Ah, sorry then, I might have been confused all along on the purpose of the `calibration_devel` branch, I thought that it was, well, something calibration-specific exclusively. I will recompile everything and let you know if it all works out. Thanks a lot for the help

aatb-ch, 2022-03-07:
Hi Felix,

So I have tried the moveit launch file for ur10 in the `calibration_devel` branch, could you confirm that the steps taken are correct? I get a `std::bad_alloc` crash on launch. Wondering if I might be missing some manual config steps?

My setup (all freshly installed and built):

- Ubuntu 20.04 Server with RT kernel (from ros-realtime) on a raspberry pi 4
- Universal Robot `ur_robot_driver`
- universal_robot `calibration_devel` branch
- URSim 3.15 with the External Control URCap

launch procedure:

- `roslaunch ur_robot_driver ur10_bringup.launch robot_ip:=192.168.0.16`
- starting program on URSim with the External Control node
- `roslaunch ur10_moveit_config ur10_moveit_planning_execution.launch`

Results in:

```
process[move_group-1]: started with pid [2498]
[ WARN] [1646650718.675808366]: MoveGroup launched with ~default_planning_pipeline '' not configured in ~planning_pipelines
[ WARN] [1646650718.679426940]: Falling back to using the move_group node's namespace (deprecated Melodic behavior).
[ INFO] [1646650718.718340688]: Loading robot model 'ur10_robot'...
[ WARN] [1646650718.894482764]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/robot_description_kinematics/manipulator/kinematics_solver_attempts' from your configuration.
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
[move_group-1] process has died [pid 2498, exit code -6, cmd /home/ubuntu/catkin_ws/devel/lib/moveit_ros_move_group/move_group __name:=move_group __log:=/home/ubuntu/.ros/log/323fc09e-9e02-11ec-95ac-11d951d09c92/move_group-1.log].
log file: /home/ubuntu/.ros/log/323fc09e-9e02-11ec-95ac-11d951d09c92/move_group-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

I've been piecing the instructions from the past `ur_bringup` procedure, replaced with the new `ur_robot_driver`. As there is no display the demo.launch expectedly crashes since it cannot start RVIZ, but the `ur10_moveit_planning_execution.launch` should run as-is?

aatb-ch, 2022-03-07:
Alright so I think I've figured out the culprit here, unsurprisingly it was me.

While following the MoveIt Servo tutorial I installed from source MoveIt, but it is already installed as a dependency of the UR driver, and since I had never used it before I did not realize that. I just reinstalled a clean Noetic install with only https://github.com/UniversalRobots/Universal_Robots_ROS_Driver and your `calibration_devel` branch and indeed the `ur10_moveit_planning_execution.launch` file starts without problem.

fmauch, 2022-07-21:
As the current version seems to be working fine on noetic, I'll close this issue. Feel free to comment / reopen if required.

closed 2022-07-21 by fmauch

gonzalocasas, 2022-07-23:
I might be wrong here but it looks to me like `melodic-devel` had the last commit in 2019 and most likely doesn't work properly on Noetic.

I assume you close it because one of the other branches or even forks of this repo works on noetic (i can confirm @fmauch 's fork does), but how is that going to be clear for anyone coming to this project without having to read a few thousand comments on issues and readmes spread all over the place...?