ros-industrial/universal_robot #539: Documentation: ur_description package structure

Victorlouisdg, 2020-09-24:
Hi, I was trying to understand the structure of the ur_description package so I made this diagram to help. I thought it might be useful for other people so maybe we could add this to a readme file that's placed inside the ur_description folder?

[image]

gavanderhoorn, 2020-09-25:
Thanks. This would indeed make a nice addition.

Note though that the structure may change, so we might have to update the diagram a few times.

hariharan382, 2020-11-04:
one doubt: in universal_robot folder there are urX_moveit_config(X be some numer) is it moveit config folder or we should generate the moveit configuration file through moveit "setup_assistant"
please answer this
asap

gavanderhoorn, 2020-11-04:
> one doubt: in universal_robot folder there are urX_moveit_config(X be some numer) is it moveit config folder or we should generate the moveit configuration file through moveit "setup_assistant"

The packages you refer to are indeed MoveIt configuration packages. I would have expected the `moveit_config` part in the name to be clear indication of that (it's customary to name MoveIt configuration pkgs for specific robots like thi).

> please answer this
> asap

Please don't do this. We're all volunteers here, helping a community in our free time.

I'm going to hide your (and my) comment, as they are off-topic for this issue. That won't remove them, only not show them by default.

hariharan382, 2020-11-05:
then i generated a package in moveit but it is completely different from
already moveit config in your universal_robot package

gavanderhoorn, 2020-11-05:
Yes, that's why I wrote you shouldn't do that, unless you have experience and would be able to edit the new configuration to work with the files we have here in this repository.

In any case: this is still off-topic for this issue. Please keep your discussion confined to #549.

hariharan382, 2020-11-06:
mam please i just want to move the bot  by slightly by python node  in
rviz,just help me for that

simon248, 2021-09-29:
You can assign it to me.