ros-industrial/universal_robot #541: Could not load controler 'arm controller' using UR3

ToonMium, 2020-10-03:
I try to launch a gazebo with UR3. Gazebo open but my robot don't move and can't be controlled.

I tried this : roslaunch ur_gazebo ur3.launch

I 've got this error message about "Arm controller" that can't be load because of the type. How can I fix it please?

[image]

When looking for rostopic list, my arm controller doesn't appear.

I 
[image]

srediske, 2020-10-03:
Your system could not start `arm_controller` because it can't find the `JointTrajectoryController` which is loaded [here](https://github.com/ros-industrial/universal_robot/blob/06d8b9e2f5f86aa54f9f2845f11edbc84e2f951e/ur_gazebo/controller/arm_controller_ur3.yaml#L2).

Make sure you have [ros_controllers](http://wiki.ros.org/ros_controllers?distro=kinetic) installed.

gavanderhoorn, 2020-10-14:
Assuming @srediske's comment solved this, I'm closing the issue.

If it wasn't solved, please update.

Ideally though you'd post this on ROS Answers @ToonMium, as this a question about configuration, not an issue with the packages in this repository.

closed 2020-10-14 by gavanderhoorn