ros-industrial/universal_robot #543: Action client not connected: /follow_joint_trajectory

barrygxwan, 2020-10-09:
After launch real ur5 robot using command:   roslaunch ur_bringup ur5_bringup.launch limited:=true robot_ip:=192.168.1.145;
I use command:  roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch limited:=true
then appears :+1: 
...
[ INFO] [1602251287.816176556]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1602251287.816187677]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1602251287.816206406]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1602251287.816222937]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1602251287.816238583]: Using planning request adapter 'Fix Start State Path Constraints'
[ WARN] [1602251292.833490319]: Waiting for /follow_joint_trajectory to come up
[ WARN] [1602251298.833842453]: Waiting for /follow_joint_trajectory to come up
[ERROR] [1602251304.834119746]: Action client not connected: /follow_joint_trajectory
[ INFO] [1602251304.860453755]: Returned 0 controllers in list
[ INFO] [1602251304.893668247]: Trajectory execution is managing controllers
...

So, How to fix it? any help will be appreciated

gavanderhoorn, 2020-10-09:
Please tell us what version controller you have: a CB2 or a CB3?

barrygxwan, 2020-10-09:
Thank you for your reply. CB3UR5 with software version:3.3.4.310

gavanderhoorn, 2020-10-09:
Then you should not be using `ur_bringup`.

Please use the correct driver: [UniversalRobots/Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver).

closed 2020-10-09 by gavanderhoorn

barrygxwan, 2020-10-09:
ok, thank you very much!

gavanderhoorn, 2020-10-09:
Note this is also clearly mentioned on [wiki/ur_driver](http://wiki.ros.org/ur_driver), and [wiki/ur_modern_driver](http://wiki.ros.org/ur_modern_driver).