ros-industrial/universal_robot #547: UR5e velocity limits in the URDF

Le-Li-AgileRobots, 2020-10-26:
Hi guys,

we were using UR5e robots and found something confusing. 

The velocity limits of the last three joints are set to `6.28 (rad/s)`, see https://github.com/ros-industrial/universal_robot/blob/06d8b9e2f5f86aa54f9f2845f11edbc84e2f951e/ur_e_description/urdf/ur5e.urdf.xacro#L238,

while the velocity limits specified in the user manual are `3.14 (rad/s)`, see the manual: https://s3-eu-west-1.amazonaws.com/ur-support-site/77147/99404_UR5e_User_Manual_en_Global.pdf

Could you guys please double-check which value is the correct one?

Many thanks.

Le Li

gavanderhoorn, 2020-10-26:
This has already been corrected in `melodic-devel-staging`. Please see #513.

gavanderhoorn, 2020-10-26:
And the follow-up #533, which uses degrees/sec for the velocity limits. Even easier to check.

gavanderhoorn, 2020-10-26:
Assuming this answers the question, I'm going to close the issue.

Feel free to keep commenting of course.

closed 2020-10-26 by gavanderhoorn

Le-Li-AgileRobots, 2020-10-27:
> Assuming this answers the question, I'm going to close the issue.
> 
> Feel free to keep commenting of course.

Yes, it answered our question.

Maybe one more comments regarding the moveit config file for UR5e in the `melodic-devel-staging` branch, see: https://github.com/ros-industrial/universal_robot/blob/e8234318cc94dd287617fc196c1b32bf6e4074a9/ur5_e_moveit_config/config/joint_limits.yaml#L22,
it seems it has not been updated with the velocity limits `PI`. 

Best,
Le Li

gavanderhoorn, 2020-10-27:
Yes, we are aware of this.

With #516, we won't be using different files for these limits any more. So that will go away as well.