ros-industrial/universal_robot #549: unable to launch ur5 in gazebo and moveit

hariharan382, 2020-11-04:
i generated moveit configuration file of ur5,but iam unable to launch the gazebo but rviz working well
`├── CMakeLists.txt
├── config
│   ├── chomp_planning.yaml
│   ├── controllers.yaml
│   ├── fake_controllers.yaml
│   ├── joint_limits.yaml
│   ├── joint_names.yaml
│   ├── kinematics.yaml
│   ├── manipulator_control.yaml
│   ├── ompl_planning.yaml
│   ├── ros_controllers.yaml
│   ├── sensors_3d.yaml
│   └── ur5.srdf
├── launch
│   ├── chomp_planning_pipeline.launch.xml
│   ├── default_warehouse_db.launch
│   ├── demo_gazebo.launch
│   ├── demo.launch
│   ├── demo_planning_execution.launch
│   ├── fake_moveit_controller_manager.launch.xml
│   ├── gazebo.launch
│   ├── joystick_control.launch
│   ├── move_group.launch
│   ├── moveit.rviz
│   ├── moveit_rviz.launch
│   ├── ompl_planning_pipeline.launch.xml
│   ├── planning_context.launch
│   ├── planning_pipeline.launch.xml
│   ├── ros_controllers.launch
│   ├── run_benchmark_ompl.launch
│   ├── sensor_manager.launch.xml
│   ├── setup_assistant.launch
│   ├── trajectory_execution.launch.xml
│   ├── trial_control.launch
│   ├── ur5_moveit_controller_manager.launch
│   ├── ur5_moveit_planning_execution.launch
│   ├── ur5_moveit_sensor_manager.launch.xml
│   ├── warehouse.launch
│   └── warehouse_settings.launch.xml
└── package.xml
`

` roslaunch demo_moveit_config demo_gazebo.launch
... logging to /home/hari/.ros/log/ba49fd2c-1ebc-11eb-9c11-080027543667/roslaunch-hari18-5706.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

RLException: while processing /home/hari/ur10_ws/src/demo_moveit_config/launch/move_group.launch:
while processing /home/hari/ur10_ws/src/demo_moveit_config/launch/trajectory_execution.launch.xml:
while processing /home/hari/ur10_ws/src/demo_moveit_config/launch/ur5_moveit_controller_manager.launch.xml:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/hari/ur10_ws/src/demo_moveit_config/launch/ur5_moveit_controller_manager.launch.xml'
The traceback for the exception was written to the log file`

### this below is the code of ur5_moveit_controller_manager.launch.xml
`<launch>

  <rosparam file="$(find demo_moveit_config)/config/controllers.yaml"/>

  <param name="use_controller_manager" value="false"/>

  <param name="trajectory_execution/execution_duration_monitoring" value="false"/>

  <param name="moveit_controller_manager" value="moveit_simple_controller_manager/MoveitSimpleControllerManager"/>

</launch> `

hariharan382, 2020-11-04:
<launch>
  <rosparam file="$(find demo_moveit_config)/config/controllers.yaml"/>
  <param name="use_controller_manager" value="false"/>
  <param name="trajectory_execution/execution_duration_monitoring" value="false"/>
  <param name="moveit_controller_manager" value="moveit_simple_controller_manager/MoveitSimpleControllerManager"/>
</launch>

hariharan382, 2020-11-04:
`<launch>
  <rosparam file="$(find demo_moveit_config)/config/controllers.yaml"/>
  <param name="use_controller_manager" value="false"/>
  <param name="trajectory_execution/execution_duration_monitoring" value="false"/>
  <param name="moveit_controller_manager" value="moveit_simple_controller_manager/MoveitSimpleControllerManager"/>
</launch>`

hariharan382, 2020-11-04:
`launch>
  <rosparam file="$(find demo_moveit_config)/config/controllers.yaml"/>
  <param name="use_controller_manager" value="false"/>
  <param name="trajectory_execution/execution_duration_monitoring" value="false"/>
  <param name="moveit_controller_manager" value="moveit_simple_controller_manager/MoveitSimpleControllerManager"/>
</launch`

hariharan382, 2020-11-04:
- <launch>
 - <rosparam file="$(find demo_moveit_config)/config/controllers.yaml"/>
 - <param name="use_controller_manager" value="false"/>
 - <param name="trajectory_execution/execution_duration_monitoring" value="false"/>
 - <param name="moveit_controller_manager" value="moveit_simple_controller_manager/MoveitSimpleControllerManager"/>
-</launch>

hariharan382, 2020-11-04:
_launch
  rosparam file="$(find demo_moveit_config)/config/controllers.yaml"/
  param name="use_controller_manager" value="false"/
  param name="trajectory_execution/execution_duration_monitoring" value="false"/
  param name="moveit_controller_manager" value="moveit_simple_controller_manager/MoveitSimpleControllerManager"/
/launch

gavanderhoorn, 2020-11-04:
Not being able to use the files generated by the MSA would be something to report over at `ros-planning/moveit`.

Unfortunately we cannot support you there. The MSA makes quite a few assumptions about structure files which may not hold true for what we do here.

The MoveIt configurations in `kinetic-devel` should still work, despite some warnings being printed.

If you're new to all of this, I would not recommend you try to generate your own, as it will require some insight into how packages and files are structured.

gavanderhoorn, 2020-11-04:
If the instructions under *Usage with Gazebo Simulation* in [the readme](https://github.com/ros-industrial/universal_robot/blob/06d8b9e2f5f86aa54f9f2845f11edbc84e2f951e/README.md), that would be something to investigate.

Have you tried those?

hariharan382, 2020-11-05:
sir i went for newly generated package since i could not run test_move.py
i am just getting "waiting for server"
i even changed linr 94 also
please help me this error toiling me for 2 weeks

ipa-nhg, 2020-11-05:
is this the same ticket than https://github.com/ros-industrial/universal_robot/issues/550 ? please don't duplicate issues and close one of them.

The python script that you want to run has nothing to do with Moveit and its configuration, this script shows an example about how to command a joint trajectory using python code.

The first question (I also told you per mail) is if the simulation was started successfully, which means if the terminal where you call the command `roslaunch ur_gazebo urX.launch limited:=true ` shows errors. Please review the print of this command.

A simple test you can do to check if the controllers were started is to run the `rqt_joint_trajectory_contrroller` (rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller) and try to move with this [simple gui](http://wiki.ros.org/rqt_joint_trajectory_controller) then joints of the robot.

hariharan382, 2020-11-06:
but there is no folder called "rqt_joint_trajectory_controller " to rosrun it,can you say how to execute the command
"rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller"

hariharan382, 2020-11-06:
`<?xml version="1.0"?>
<launch>
  <arg name="limited" default="false" doc="If true, limits joint range [-PI, PI] on all joints." />
  <arg name="transmission_hw_interface" default="hardware_interface/PositionJointInterface" />

  <param unless="$(arg limited)" name="robot_description" command="$(find xacro)/xacro '$(find ur_description)/urdf/ur5_robot.urdf.xacro' transmission_hw_interface:=$(arg transmission_hw_interface)" />
  <param if="$(arg limited)" name="robot_description" command="$(find xacro)/xacro '$(find ur_description)/urdf/ur5_joint_limited_robot.urdf.xacro' transmission_hw_interface:=$(arg transmission_hw_interface)" />
</launch>`

ur5_upload.launch

> roslaunch ur_robot_driver ur5_bringup.launch
... logging to /home/hari/.ros/log/eff511f0-1ff7-11eb-ae0f-080027543667/roslaunch-hari18-11443.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

RLException: unused args [kinematics_config] for include of [/home/hari/ur10_ws/src/universal_robot/ur_description/launch/ur5_upload.launch]
The traceback for the exception was written to the log file

hariharan382, 2020-11-06:
i am getting this above error ,when i launch roslaunch ur_robot_driver ur5_bringup.launch

hariharan382, 2020-11-06:
i just want to move the bot by python node in rviz which i am trying for nearly a month,facing different errors each time i try to run it

ipa-nhg, 2020-11-06:
> but there is no folder called "rqt_joint_trajectory_controller " to rosrun it,can you say how to execute the command
> "rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller"

`sudo apt install ros-$ROS_DISTRO-rqt-joint-trajectory-controller ` 

> i am getting this above error ,when i launch roslaunch ur_robot_driver ur5_bringup.launch

This is not starting the gazebo simulation for this robot, also the package `ur_robot_driver` is not part of this repository. I will close the issue you should report this to the right repository.

Hint: I think you are using the wrong version of the `ur_description` package, see the instructions of the `Universal_Robots_ROS_Driver` https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#building , the driver you are launching.

closed 2020-11-06 by ipa-nhg

gavanderhoorn, 2020-11-06:
Let's get a base-line.

@hariharan382: if you follow the instructions in the readme, *verbatim*, for starting Gazebo, MoveIt and RViz. Does that work?

If not, please report any errors or warnings you see. And *always* show the commands you run.

gavanderhoorn, 2020-11-06:
Do *not* edit any files. Let's first try to get things working as they are.

I'm not sure also *why* you've started to edit files. There should be no need.

At least not to get the basics working.

gavanderhoorn, 2020-11-06:
And I'm going to repeat my previous request:

> And please try to use Github [code blocks](https://docs.github.com/en/free-pro-team@latest/github/writing-on-github/creating-and-highlighting-code-blocks) when copy-pasting console text and/or code.

if you don't use codeblocks, whatever you copy-paste from the terminal is difficult to read.

hariharan382, 2020-11-06:
sir i could see the ur10 in gazebo and rviz but i just want to move it with test_move.py
please help me with that when ever i try i am getting 
waiting for server(i know it is controller error  but "rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller")
unable to launch it even i could not find rqt_joint_trajectory_controller in Universal_Robots_ROS_Driver  please check your self

please just dont close issue without providing solutions

hariharan382, 2020-11-06:
i will show tree of Universal_Robots_ROS_Driver can you show me rqt_joint_trajectory_controller folder

hariharan382, 2020-11-06:
user:~$rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller
[rospack] Error: package 'rqt_joint_trajectory_controller' not found

/home/hari/ur10_ws/src/Universal_Robots_ROS_Driver
├── controller_stopper
│   ├── include
│   │   └── controller_stopper
│   └── src
├── ur_calibration
│   ├── include
│   │   └── ur_calibration
│   ├── launch
│   ├── src
│   └── test
├── ur_controllers
│   ├── doc
│   ├── include
│   │   └── ur_controllers
│   └── src
├── ur_dashboard_msgs
│   ├── action
│   ├── msg
│   └── srv
└── ur_robot_driver
    ├── config
    ├── doc
    │   └── initial_setup_images
    ├── include
    │   └── ur_robot_driver
    ├── launch
    ├── resources
    ├── scripts
    ├── src
    └── test

hariharan382, 2020-11-06:
can you find where is rqt file
?

gavanderhoorn, 2020-11-06:
First: please try to post *single* comments here. Do not put each sentence in its own comment.

> unable to launch it even i could not find rqt_joint_trajectory_controller in Universal_Robots_ROS_Driver

`rqt_joint_trajectory_controller` is a *package*, it is not a folder in `Universal_Robots_ROS_Driver`.

It is a plugin for RQT. See [wiki/rqt_joint_trajectory_controller](http://wiki.ros.org/rqt_joint_trajectory_controller).

So you cannot "find it" in `Universal_Robots_ROS_Driver`, as it is not there.

If you don't have this plugin installed, then you cannot `rosrun` it. So please make sure you have `rqt_joint_trajectory_controller` installed (note: it will be called `ros-kinetic-rqt-joint-trajectory-controller` or `ros-melodic-rqt-joint-trajectory-controller`, depending on which version of ROS you are using). Do **not** build it from source in your Catkin workspace.

hariharan382, 2020-11-06:
then can you say where it is so that i can run

hariharan382, 2020-11-06:
sudo apt install ros-$ROS_melodic-rqt-joint-trajectory-controller
[sudo] password for hari: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros--rqt-joint-trajectory-controller

ipa-nhg, 2020-11-06:
`sudo apt install ros-$ROS_DISTRO-rqt-joint-trajectory-controller ` as I said [here](https://github.com/ros-industrial/universal_robot/issues/549#issuecomment-722961295)

or 

`sudo apt install ros-melodic-rqt-joint-trajectory-controller` as gavanderhoorn said [here](https://github.com/ros-industrial/universal_robot/issues/549#issuecomment-723081778)

see the [basic ROS documentation](http://wiki.ros.org/melodic/Installation/Ubuntu) about how to install single packages for further information.

hariharan382, 2020-11-06:
'''hari@hari18:~$ rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller
qt_gui_main() found no plugin matching "rqt_joint_trajectory_controller"
try passing the option "--force-discover"'''

ipa-nhg, 2020-11-09:
did you try with the `--force-discover` option? If yes and didn't work, you should check your installation and the source of the ROS environment. See if the $ROS_PACKAGE_PATH env variable includes the folder where you installed the ROS package (likely /opt/ros/melodic/share) 

You  can also try the command `roscd rqt_joint_trajectory_controller`