ros-industrial/universal_robot #554: Can you add urdf files to universal_robot/ur_description/urdf/

infinitespace, 2020-11-20:
Hi Universal Robot experts,

Do you mind adding urdf file into `universal_robot/ur_description/urdf/`? Otherwise users have to install xacro and ROS.

Simple upload urdf file would be much appreciated since many researchers and robot algorithm developers are using simulations without ROS or xacro installed.

Thank you very much.

gavanderhoorn, 2020-11-20:
This would seem to need a revert of #209. The rationale for *that* PR was (quoting @fmessmer):

> I really would like to get rid of these files...
> ...it is a pain maintaining them
> ...they are nowhere used within this repo
> ...they can easily be created
> ...it's getting too many files (-> introducing_ur3)

for the case you describe (using these files without ROS), only point 3 seems to not apply.

What about we come up with a way to convert the `.xacro` files without having to install ROS? Would that work?

I'd really like to avoid storing `.urdf`s here for the reasons mentioned.

gavanderhoorn, 2020-11-20:
Additionally: with the addition of support for kinematics calibration (in `Universal_Robot_ROS_Driver`), it would seem impossible to provide `.urdf`s here, as there would be no way to update them based on the exact kinematics of *your* robot.

That's one reason why we have to use xacro here.

gavanderhoorn, 2020-11-26:
I'm going to close this due to inactivity.

closed 2020-11-26 by gavanderhoorn