ros-industrial/universal_robot #555: Are there any descriptions about these packages?

Steven-ZY, 2020-11-24:
Hello, everyone. I'm a fresh guy. Are there any descriptions about these packages? Such as their functions , the difference between `urx_e_moveit_congfig` and `urx_moveit_congfig`

gavanderhoorn, 2020-11-24:
Each package always contains a manifest (ie: `package.xml`) which should contain a `description` element with some descriptive text.

Other than that: no, we don't have any additional explanation about these packages afaik.

Gold plating that a bit: the packages in this repository (try to) follow ROS best practices wrt naming and functionality. Besides allowing users to reuse their experience with other packages, this also allows us to *lean* on that experience a bit. One thing this gets us is that instead of having to provide documentation for each and every individual package, we can just say "this is a description package" and "this is a MoveIt configuration". The assumption is then that users would know what "a description package" is, and thus be able to work with whatever we offer here.

I'm not claiming that's a perfect situation though -- especially not when you just start out.

If you're willing to help future users by providing some more description we can certainly merge that.

I'd add it to the/a readme per package.

gavanderhoorn, 2020-11-24:
Oh and btw: if you are willing to contribute such a description, please open the PRs against `melodic-devel-staging`, as that's where the focus of development is.

gavanderhoorn, 2021-01-29:
I'm going to close this due to inactivity.

if this is still an issue, please comment and re-open.

And as always: feel free to keep commenting on this issue.

closed 2021-01-29 by gavanderhoorn