ros-industrial/universal_robot #556: Saving the URDF file for a specific robot

Victorlouisdg, 2020-12-02:
Hi, I'm wondering the easiest way is to save the URDF file for a specific robot? Something like this would be convenient, but I guess it's not possible:
`roslaunch ur_description load_ur10.launch > my_ur_robot.urdf`

I also tried directly using the `xacro` command, but I can't find examples of how to pass arguments:
`xacro ur.xacro > my_ur_robot.urdf`

Victorlouisdg, 2020-12-02:
My current workaround is to copy the `ur.xacro` file and manually fill in the arguments, but I'm sure there must be a better way :p

fmauch, 2020-12-02:
One way would be

```
rosrun xacro xacro $(rospack find ur_description)/urdf/ur.xacro joint_limit_params:=$(rospack find ur_description)/config/ur10/joint_limits.yaml kinematics_params:=$(rospack find ur_description)/config/ur10/default_kinematics.yaml physical_params:=$(rospack find ur_description)/config/ur10/physical_parameters.yaml visual_params:=$(rospack find ur_description)/config/ur10/visual_parameters.yaml
```

@gavanderhoorn I remember in the back of my head that we though about having one `.xacro` file per robot type that users could use to build their own environment descriptions. This would have the defaults for the robot types already inserted. Do I remember correctly? This might also be quite nice for #538 as we could have robots named `urX` instead of just `ur`.

gavanderhoorn, 2020-12-02:
@Victorlouisdg wrote:
> I guess it's not possible:
> 
> `roslaunch ur_description load_ur10.launch > my_ur_robot.urdf`

No, not like that.

What would be possible would be to use `rosparam get /robot_description > my_ur_robot.urdf` after you ran `roslaunch ur_description load_ur10.launch`. It won't be pretty, but should be valid URDF.

I haven't tested this though.

@fmauch wrote:
> I remember in the back of my head that we though about having one `.xacro` file per robot type that users could use to build their own environment descriptions

Yes, that is indeed what we've discussed. Basically this: https://github.com/gavanderhoorn/universal_robot/commit/4187b12e7318d626cc179d634094214c9a8368b9.

gavanderhoorn, 2021-01-29:
I'm going to close this, as I believe we've addressed the question in the OP.

Feel free to keep commenting on the issue of course.

closed 2021-01-29 by gavanderhoorn