ros-industrial/universal_robot #558: RobotState No msg received

Csp0920, 2020-12-22:
Here is a problem that I have ignored.
I am using ros-melodic.
When I use the package "universal_robot" and package "fmauch_universal_robot ",  RobotState in moveit show that "No msg received".
However, It seems that no error will be caused.
So if this problem need to be focused?

ipa-nhg, 2021-01-15:
@Csp0920 Thanks for finding this issue.

What do you mean with `fmauch_universal_robot`? Which version of `[universal_robot](https://github.com/ros-industrial/universal_robot)` are you using?

Csp0920, 2021-01-18:
> @Csp0920 Thanks for finding this issue.
> 
> What do you mean with `fmauch_universal_robot`? Which version of `[universal_robot](https://github.com/ros-industrial/universal_robot)` are you using?

It's from  `https://github.com/fmauch/universal_robot.git src/fmauch_universal_robot`, and the version is calibration_devel.

ipa-nhg, 2021-01-18:
I am sorry, but I ignore all the differences between the upstream version and the fork/branch that you are using. 

As general hint: likely there is just an inconsistency between the name of the topic expected by Moveit! and the one published by the driver, and you need only to remap interfaces. You can use the command `rostopic list` to introspect it.

gavanderhoorn, 2021-01-18:
I don't believe we have sufficient information to help in this case.

`RobotState` is a MoveIt display.

It's possible you're not starting everything correctly, or should be using the [Robot Model display](http://wiki.ros.org/rviz/DisplayTypes/RobotModel) instead if not using MoveIt.

See #560 for a discussion on what is needed to successfully use MoveIt with `Universal_Robots_ROS_Driver`.

gavanderhoorn, 2021-01-24:
I'm going to close this.

If it's still a problem for you @Csp0920, please comment here and provide more information.

closed 2021-01-24 by gavanderhoorn