ros-industrial/universal_robot #559: ERROR: GazeboRosControlPlugin missing <legacyModeNs>

gio-hash, 2020-12-22:
Hi, when i try to launch the ur5 robot in gazebo with this command: roslaunch ur_gazebo ur5.launch , in the upload of the robot simulation give me these error : {GazeboRosControlPlugin missing <legacyModeNS> while using DefaultRobotHWSim, defaults to true.
This setting assumes you have an old package with an old implementation of DefaultRobotHWSim, where the robotNamespace is disregarded and absolute paths are used instead.
If you do not want to fix this issue in an old package just set <legacyModeNS> to true.}
and i can not try to solve it.Can someone helpe me?
thanks Giovanni

ipa-nhg, 2021-01-15:
@gio-hash thanks for the report.

Could you please give us some details about the ROS distro and the version of the [universal_robot](https://github.com/ros-industrial/universal_robot.) repository you installed?

gio-hash, 2021-01-15:
> @gio-hash thanks for the report.
> 
> Could you please give us some details about the ROS distro and the version of the [universal_robot](https://github.com/ros-industrial/universal_robot.) repository you installed?

Hi i use ros-kinetic on ubuntu 16.04 and i have installed the latest version of the universal robot

ipa-nhg, 2021-01-18:
Try uncommentating the following line:

https://github.com/ros-industrial/universal_robot/blob/06d8b9e2f5f86aa54f9f2845f11edbc84e2f951e/ur_description/urdf/common.gazebo.xacro#L7

gavanderhoorn, 2021-01-24:
I'm going to close this.

I'm not sure, but I don't believe the message @gio-hash reports is actually an error, or if it is, it doesn't influence the behaviour of the robot in Gazebo (there are other more serious problems with that).

If you do feel this is a problem @gio-hash, please post an update as a comment here and would re-open if necessary.

closed 2021-01-24 by gavanderhoorn