ros-industrial/universal_robot #560: Compatibility of medolic-devel-staging with UR ROS Driver

Victorlouisdg, 2021-01-05:
Hi I was wondering whether the `melodic-devel-staging` branch of this repository is already compatible with any branch from the [UR ROS Driver repository](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver)? 

I'm either having trouble installing the right dependencies or when I use the [fmauch's fork of this repository](https://github.com/fmauch/universal_robot) I can't seem to use my UR3e robot with MoveIt correctly. When I bringup the robot and then run:

`roslaunch ur3_moveit_config moveit_rviz.launch config:=true` 

RViz launches and I can see the robot, but it's in the wrong position. In the terminal I also get this error:

`[ERROR] [1609855066.783030367]: Semantic description is not specified for the same robot as the URDF`

Which is probably because I'm using the ur3_moveit_config instead of one for the UR3e. 

Any help/pointers are appreciated :)

Victorlouisdg, 2021-01-05:
To get the rosdep working I did the following steps:

1. clone both the UR ROS Driver repository and fmauch's fork of this repository
2. rosdep install --from-paths src --ignore-src -y
3. remove fmauch's branch
4. clone melodic-devel-staging
5. catkin_make

The catkin_make is successful so that's good. However, bringing up the robot with:
`roslaunch ur_robot_driver ur3e_bringup.launch robot_ip:=10.42.0.162`
Fails with the errors that the `ur_e_description` package can't be found, which indeed doesn't exist.

I guess this might be fixable just with new launch files. Has anyone started working on that?

gavanderhoorn, 2021-01-05:
I don't believe you can use `melodic-devel-staging` as-is with `ur_robot_driver` right now.

One of the main incompatibilities is the way `ur_robot_driver` (and `fmauch`'s fork) stores/encodes calibration data. I believe axis orientation is different among other things, so afaik you can't substitute one for the other.

> `[ERROR] [1609855066.783030367]: Semantic description is not specified for the same robot as the URDF`
> 
> Which is probably because I'm using the ur3_moveit_config instead of one for the UR3e.

yes, that's most likely the reason.

But [fmauch/universal_robot@calibration_devel](https://github.com/fmauch/universal_robot/tree/calibration_devel) is based on `melodic-devel-staging` partly, and that branch contains a `ur_e_description` and `ur3_e_moveit_config`. Can you clarify why you don't use those?

Victorlouisdg, 2021-01-05:
Thanks for the quick response! :)

My bad, I was confused, was using the wrong branch of fmauch's fork which didn't have the ur3_e_moveit_config. However I've reinstalled (with the correct branch) and strangely I'm still getting the same error.

After running:
`roslaunch ur_robot_driver ur3e_bringup.launch robot_ip:=10.42.0.162`
`roslaunch ur3_e_moveit_config demo.launch`
RViz opens with the robot in the wrong position, and these errors:
`[ERROR] [1609862335.023538487]: Semantic description is not specified for the same robot as the URDF`
`[ERROR] [1609862331.794210356]: Exception while loading move_group capability 'move_group/MoveGroupExecuteService': MultiLibraryClassLoader: Could not create object of class type move_group::MoveGroupExecuteService as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()`

Victorlouisdg, 2021-01-05:
Upon inspection I just realized `demo.launch` is not meant to be used with a real robot.

Victorlouisdg, 2021-01-05:
When running:
` roslaunch ur3_e_moveit_config moveit_rviz.launch config:=true`
It get a slightly different error:
`[ERROR] [1609863591.755269311]: Robot semantic description not found. Did you forget to define or remap '/robot_description_semantic'?`

The robot also still appears in the wrong pose. I've also noticed that all published joint states are perfectly 0, so maybe part of the issue lies there. The bringup seems to work fine, but my robot is in an upright position, where the joint states shouldn't be zero I guess.

gavanderhoorn, 2021-01-05:
> Upon inspection I just realized `demo.launch` is not meant to be used with a real robot.

No, indeed.

You have to run more than just a single `roslaunch` command.

The readme shows the files you have to `.launch`. See the *MoveIt! with real Hardware* section.

of course don't run the `ur_bringup` command.

> The robot also still appears in the wrong pose. I've also noticed that all published joint states are perfectly 0, so maybe part of the issue lies there. 

please make sure you don't have an instance of the `joint_state_publisher` still running somewhere.

Victorlouisdg, 2021-01-05:
So if I understand correctly I have to run these commands: 
```
roslaunch ur_robot_driver ur3e_bringup.launch robot_ip:=10.42.0.162
roslaunch ur3e_moveit_config ur3e_moveit_planning_execution.launch
roslaunch ur3_e_moveit_config moveit_rviz.launch config:=true
```
And indeed the when running the second command I get:
```
[ERROR] [1609865675.498039261]: Action client not connected: /follow_joint_trajectory
```
I'm not sure what exactly to fill into the `controllers.yaml` file though. This?:
``` 
  - name: "scaled_pos_joint_traj_controller"
```
Thanks already for the help! I'll be trying to get this to work again tomorrow :)

Victorlouisdg, 2021-01-06:
I figured out why all joint states were 0, I hadn't powered the robot ON in Polyscope... Everything seems to be working fine now :)

closed 2021-01-06 by Victorlouisdg