ros-industrial/universal_robot #561: [Question] How to retrieve all IK solutions using ur_kinematics package

Thieso, 2021-01-20:
Hi guys, 

I am working with a UR10e robot and am planning a path in cartesian space. I the start and end points are close to each other it works mostly fine but I they are rather far away I have a lot of issues with the inverse kinematics solutions. It keeps finding very extreme positions (some joints close to their limits) and if I add the floor as an obstacle to avoid those solutions, it does not find a solution at all. I already merged #358 but that did not help. 
So my idea was to retrieve all IK solutions and choose one based on my own criteria (something like minimum distance in joint space with some weights on the joints). 
My question is now, how can I retrieve these solutions from my Moveit Package to do these weighting? 

I am not sure if this is the right place to ask such questions but I would appreciate any help. 

Thanks a lot.

gavanderhoorn, 2021-01-20:
If you're using MoveIt, I would recommend to ask this on the MoveIt issue tracker.

All we can do here is provide the IK plugin, which you apparently already have.

Thieso, 2021-01-20:
Ok sure I will ask there then. Thanks for the quick reply.

closed 2021-01-20 by Thieso