ros-industrial/universal_robot #563: Semantic description is not specified for the same robot as the URDF

keke-220, 2021-01-29:
I am trying to run ur5e on gazebo. When I run "roslaunch ur5_e_moveit_config ur5_e_moveit_planning_execution.launch limited:=true", it shows some errors:

[ERROR] [1611882469.871418732]: Semantic description is not specified for the same robot as the URDF
[ERROR] [1611882469.872031288]: Link 'ee_link' declared as part of a chain in group 'manipulator' is not known to the URDF
[ERROR] [1611882469.872057281]: Link 'ee_link' declared as part of group 'endeffector' is not known to the URDF
[ERROR] [1611882469.872129461]: Link 'ee_link' specified as parent for end effector 'moveit_ee' is not known to the URDF
[ERROR] [1611882469.898956814]: Group state 'home' specified for group 'manipulator', but that group does not exist
[ERROR] [1611882469.898964819]: Group state 'up' specified for group 'manipulator', but that group does not exist
[ERROR] [1611882469.946106525]: Failed to find 3D sensor plugin parameters for octomap generation

Does anyone know how it deal with it? Thanks!

SUMMARY
========

PARAMETERS
 * /move_group/allow_trajectory_execution: True
 * /move_group/capabilities: move_group/MoveGr...
 * /move_group/controller_list: [{'action_ns': 'f...
 * /move_group/endeffector/planner_configs: ['SBLkConfigDefau...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/manipulator/default_planner_config: RRTConnectkConfig...
 * /move_group/manipulator/longest_valid_segment_fraction: 0.01
 * /move_group/manipulator/planner_configs: ['SBLkConfigDefau...
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/ESTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/ESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/KPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/KPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/KPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/KPIECEkConfigDefault/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/LBKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/LBKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/LBKPIECEkConfigDefault/type: geometric::LBKPIECE
 * /move_group/planner_configs/PRMkConfigDefault/max_nearest_neighbors: 10
 * /move_group/planner_configs/PRMkConfigDefault/type: geometric::PRM
 * /move_group/planner_configs/PRMstarkConfigDefault/type: geometric::PRMstar
 * /move_group/planner_configs/RRTConnectkConfigDefault/range: 0.0
 * /move_group/planner_configs/RRTConnectkConfigDefault/type: geometric::RRTCon...
 * /move_group/planner_configs/RRTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/RRTkConfigDefault/range: 0.0
 * /move_group/planner_configs/RRTkConfigDefault/type: geometric::RRT
 * /move_group/planner_configs/RRTstarkConfigDefault/delay_collision_checking: 1
 * /move_group/planner_configs/RRTstarkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/RRTstarkConfigDefault/range: 0.0
 * /move_group/planner_configs/RRTstarkConfigDefault/type: geometric::RRTstar
 * /move_group/planner_configs/SBLkConfigDefault/range: 0.0
 * /move_group/planner_configs/SBLkConfigDefault/type: geometric::SBL
 * /move_group/planner_configs/TRRTkConfigDefault/frountierNodeRatio: 0.1
 * /move_group/planner_configs/TRRTkConfigDefault/frountier_threshold: 0.0
 * /move_group/planner_configs/TRRTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/TRRTkConfigDefault/init_temperature: 10e-6
 * /move_group/planner_configs/TRRTkConfigDefault/k_constant: 0.0
 * /move_group/planner_configs/TRRTkConfigDefault/max_states_failed: 10
 * /move_group/planner_configs/TRRTkConfigDefault/min_temperature: 10e-10
 * /move_group/planner_configs/TRRTkConfigDefault/range: 0.0
 * /move_group/planner_configs/TRRTkConfigDefault/temp_change_factor: 2.0
[... 105 lines cut ...]
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...
[ INFO] [1611882470.541863898, 719.752000000]: 

********************************************************
* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1611882470.541962713, 719.752000000]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1611882470.542005274, 719.752000000]: MoveGroup context initialization complete

You can start planning now!

keke-220, 2021-01-29:
I was running the melodic-devel-staging branch on Melodic.
And I was planning to run:
 roslaunch ur_gazebo ur5e_bringup.launch limited:=true
roslaunch ur5_e_moveit_config ur5_e_moveit_planning_execution.launch
roslaunch ur5_e_moveit_config moveit_rviz.launch config:=true
Did I miss anything?

gavanderhoorn, 2021-01-29:
There is no MoveIt support in `melodic-devel-staging`.

See #538.

gavanderhoorn, 2021-01-29:
As I believe we've identified the cause of the problem you report, I'm going to close this issue.

If it turns out that was not the cause, we can re-open.

In the meantime, feel free to keep commenting on it.

gavanderhoorn, 2021-01-29:
Finally: please note that `melodic-devel-staging` is just that, a *staging ground*. It's not a released branch at the moment, so things may -- and likely will -- break at any time.

closed 2021-01-29 by gavanderhoorn