ros-industrial/universal_robot #565: Open-source license type

sarah-gibson, 2021-02-24:
For the ROS packages in this project that specify BSD license type in package.xml, is this referring to the 3-Clause BSD license?

Also, to whom and what year can I attribute the copyright for packages in this repo?

Thanks!

gavanderhoorn, 2021-03-01:
> For the ROS packages in this project that specify BSD license type in package.xml, is this referring to the 3-Clause BSD license?

yes, BSD 3-Clause, where `BSD` is specified.

> Also, to whom and what year can I attribute the copyright for packages in this repo?

All packages should list various authors in the `author` elements in the manifests. For some packages though (such as `ur_description`), there are many more contributors than listed in the manifests, as it would be infeasible to track them all.

Example for `ur_description` again:

https://github.com/ros-industrial/universal_robot/blob/e8234318cc94dd287617fc196c1b32bf6e4074a9/ur_description/urdf/ur_macro.xacro#L33-L51

I would suggest to credit "the $pkg contributors" in those cases.

gavanderhoorn, 2021-03-01:
Assuming your questions have been answered, I'm going to close this.

Feel free to keep commenting, and if something is unclear, we can re-open.

closed 2021-03-01 by gavanderhoorn