ros-industrial/universal_robot #566: UR_Gazebo fidelity vs real hardware

tianshiz, 2021-03-10:
I saw the discussion in #521 and have some questions.
How accurate is the current UR_Gazebo simulation using hardware_interface/PositionJointInterface? Is it a good enough approximation of the dynamics of the arm if not interacting with other objects? Can I have any sense of confidence that I can get analogous behavior between the sim and real arm?

If we are switching to hardware_interface/EffortJointInterface for the gazebo controller, then would the moveit controller also need to use effort_controllers/JointTrajectoryController(otherwise how do we even compare the gazebo arm with the real)? If I tune the controller PID's in gazebo, are those values likely to have any bearings on the tuning for the real arm? 

I want a gazebo sim that can approximate my real arm performance/behavior loaded/unloaded. Since I'm using a position_controllers/JointTrajectoryController for the arm, the current ur_gazebo config matches it better. For my purposes, is the current config good enough?

gavanderhoorn, 2021-03-10:
> How accurate is the current UR_Gazebo simulation using hardware_interface/PositionJointInterface? 

Not very.

> Is it a good enough approximation of the dynamics of the arm if not interacting with other objects?

I would not depend on it -- but afaik, there have been no comparisons (as in: papers published).

> Can I have any sense of confidence that I can get analogous behavior between the sim and real arm?

no, I don't believe so.

> If we are switching to hardware_interface/EffortJointInterface for the gazebo controller, then would the moveit controller also need to use effort_controllers/JointTrajectoryController(otherwise how do we even compare the gazebo arm with the real)? 

No.

MoveIt "controllers" are not the same as `ros_control` controllers.

> If I tune the controller PID's in gazebo, are those values likely to have any bearings on the tuning for the real arm?

No.

Gazebo doesn't use anything from "the real arm", nor does "the real arm" use anything from Gazebo.

If you're asking about tuned PIDs for velocity interfaces, then I would not assume PIDs working with Gazebo will work with real hw or vice-versa.

> I want a gazebo sim that can approximate my real arm performance/behavior loaded/unloaded. Since I'm using a position_controllers/JointTrajectoryController for the arm, the current ur_gazebo config matches it better. For my purposes, is the current config good enough?

I can't really say, as there are too many variables.

One of those variables would be the version of Gazebo: behaviour of simulations changes between major versions of Gazebo, so something which works *well enough* in Gazebo on Kinetic, might not work with Melodic, but could suddenly start working again in Noetic.

gavanderhoorn, 2021-03-10:
I would be interested in a comparison of the dynamics of the arm in Gazebo versus real hw.

It's well possible someone or a group has performed a system identification of the different variants. That would get us one half of such a comparison.

What remains would be to tune the Gazebo models such that the dynamics start to approach the ones identified.

gavanderhoorn, 2021-03-10:
Finally: as this is -- right now -- not an actual issue with the packages in this repository (it's more of a question/discussion), I'm going to close it.

If it turns out there is something to fix, we can re-open -- or track those problems in new issues.

In the meantime: feel free to keep commenting on it.

closed 2021-03-10 by gavanderhoorn

tianshiz, 2021-03-10:
thanks for the quick and thorough response. 
Will moving to hardware_interface/EffortJointInterface change your answers to some of my questions(pertaining to the fidelity of the sim)?

gavanderhoorn, 2021-03-10:
Related PRs: #504, #525 and #526.

gavanderhoorn, 2021-03-11:
Is this you:

[accuracy of gazebo simulation of arm vs real hardware](https://answers.gazebosim.org/question/26675)

?

tianshiz, 2021-03-11:
yea, I wanted to pose the question to the gazebo experts for their perspective. Mainly wanted to see if anyone has done something like this in general and if they had any data and learn from their methodology. 
There was the Atlas robot done for DRC in gazebo, which was able to walk in simulation. I'm not sure if walking in gazebo matches the dynamics of the real atlas, but that would have been very helpful for that challenge. It that worked, I don't see why an arm can't work. And as you mentioned, its much more than the hardware_interface. 

While it would be amazing to spend the time to investigate the UR arm I have. I unfortunately have to focus on a different arm for this investigation. I was hoping actually that someone else had done this on the UR since it's such a popular platform.

gavanderhoorn, 2021-03-11:
> I was hoping actually that someone else had done this on the UR since it's such a popular platform.

well, have you searched for this?

I doubt *everyone* using URs watches this issue tracker. So it may well be that someone has already done the work, published about it (fi) but hasn't reported back.

Unfortunately that does happen often.