ros-industrial/universal_robot #567: Ur5 with moveit

nawrezz, 2021-04-15:
is it possible to get the same mouvement as movep with blend radius with moveit?

fmauch, 2021-05-01:
URScript's `movep` and generating a trajectory using MoveIt is quite different. 

ROS trajectories always go through the specified trajectory setpoints exactly without cutting corners as traditional blending does. Whether the robot will stop in a waypoint or how it will move around the waypoint is determined also by the velocities defined for a waypoint. Waypoints are connected to each other using splines leading to a continuous motion.

As this driver is only executing the trajectories generated by MoveIt! there's not much I can say from a driver's perspective to this. You might want to try asking on ROS Answers, but you should probably describe in more detail what you would actually like to achieve.

ipa-nhg, 2021-09-29:
@nawrezz could you please give more information? A more elaborated question will be very helpful

fmauch, 2022-07-21:
I'll remove the wrid21 label as this is nothing that can be worked on directly. I'm not sure whether this can be resolved on this level, but to further clarify this, more input would be needed.